Arksine/moonraker #722: Can not upload gcode files (check_write_enabled)

Perlovka, 2023-10-07:
### What happened

Hi, I've set up klippy, moonrake and mainsail (latest git versions)

Service parameters:
```
/usr/lib/python-exec/python3.10/python /home/klipper/klippy/klippy.py /var/lib/klipper/config/printer.cfg -a /var/lib/klipper/comms/klippy.sock -l /var/lib/klipper/logs/klippy.log
/usr/lib/python-exec/python3.10/python -m moonraker -c /var/lib/klipper/config/moonraker.cfg -d /var/lib/klipper -l /var/lib/klipper/logs/moonraker.log
```
Directory structure:
```
klipper@orange /var/lib/klipper $ tree
.
├── certs
├── comms
│   ├── klippy.sock
│   └── moonraker.sock
├── config
│   ├── moonraker.cfg
│   └── printer.cfg
├── database
│   ├── data.mdb
│   └── lock.mdb
├── gcodes
├── logs
│   ├── klippy.log
│   └── moonraker.log
└── moonraker.asvc
```

When I try to upload any gcode via mainsail, I get following error:
```
==> /var/lib/klipper/logs/moonraker.log <==
HTTPServerRequest(protocol='http', host='klipper.perlovka.net', method='POST', uri='/server/files/upload', version='HTTP/1.1', remote_ip='10.0.0.2')
Traceback (most recent call last):
  File "/home/klipper/.local/lib/python3.10/site-packages/tornado/web.py", line 1690, in _execute
    result = self.prepare()
  File "/home/klipper/.local/lib/python3.10/site-packages/moonraker/app.py", line 926, in prepare
    fm.check_write_enabled()
  File "/home/klipper/.local/lib/python3.10/site-packages/moonraker/components/file_manager/file_manager.py", line 251, in check_write_enabled
    raise self.server.error(
moonraker.utils.ServerError: Write access is currently disabled.  Check notifications for warnings.
2023-10-07 10:33:50,965 [app.py:log_request()] - 500 POST /server/files/upload (10.0.0.2) [No User] 29.34ms
```

I have no warning notifications except
```
Moonraker warning
No machine provider configured, cannot validate service file.
```

Obviosly because machine provider is set to `none`, there is no systemd installed.

moonraker.cfg:
```
[server]
host: 0.0.0.0
port: 7125
klippy_uds_address: /var/lib/klipper/klippy.sock

[machine]
provider: none
```

printer.cfg (relevant part):
```
[virtual_sdcard]
path: /var/lib/klipper/gcodes
on_error_gcode: CANCEL_PRINT
```

### Client

Mainsail

### Browser

Firefox

### How to reproduce

Try to upload gcode file via MainSail UI

Perlovka, 2023-10-07:
I finally found a problem.
Following function is called, when machine provider is set to `none`:

fm.disable_write_access() https://github.com/Arksine/moonraker/blob/9785387867605ce5962ea76262d840b9121553d4/moonraker/components/machine.py#L1552C13-L1552C38

Because of `No machine provider configured, cannot validate service file`

It's a very bad idea to tie application functionality to such bloatware like systemd.

Arksine, 2023-10-07:
As per the [documentation](https://moonraker.readthedocs.io/en/latest/configuration/#machine) you need to disable service validation by setting `validate_service: False` in the `[machine]` section of the configuration.

Moonraker's primary supported target OS Debian and its derivatives, thus the base installation assumes systemd is present and responsible for service management.  However, as mentioned above, Moonraker can be configured to run on custom linux installations.

Perlovka, 2023-10-07:
Thanks, yes, I've already found this option.
Maybe setting machine provider to `none` should also disable service files validations? Just an idea.

Anyway, I'm closing this task, since my problem is resolved.
Thank you :)

closed 2023-10-07 by Perlovka

essorblood, 2024-03-24:
How did you solve it? Can you tell me more?

Perlovka, 2024-08-19:
> How did you solve it? Can you tell me more?

@essorblood https://github.com/Arksine/moonraker/issues/722#issuecomment-1751813129