Arksine/moonraker #725: Add reset of temperature store endpoint

pedrolamas, 2023-10-13:
### Is your feature request related to a problem? Please describe

I've been seeing users requesting the ability to reset the temperature store (specifically, https://github.com/fluidd-core/fluidd/issues/1169), so I would ask for such a feature in the Moonraker API so we can use it in the interfaces.

### Describe the solution you'd like

Maybe a `DELETE /server/temperature_store` and equivalent `server.temperature_store.delete` or something of the likes of this.

### Describe alternatives you've considered

Have the frontend store a timestamp as a base limit from which to present results (we currently use this approach for the command line history)

Arksine, 2023-10-15:
The short term solution is to restart the Moonraker service, as this will clear the temperature store without requiring a reboot of the entire machine.

I'm a bit weary of adding an API to perform this action.  Generally this is data that shouldn't be modified by the user.  The optimal solution would be to fix Klipper's temperature reporting in this scenario.

pedrolamas, 2023-10-15:
Can't disagree on that! In fairness, I too would prefer a proper fix upstream in Klipper...

In any case, I might go ahead and add a timestamp limit in Fluidd, so it would only affect Fluidd and nothing else.

closed 2023-10-15 by pedrolamas