Arksine/moonraker #726: Websocket infinite reconnection on desktop web browsers, when accessing to an iPhone it works

SpaghettiC0des, 2023-10-13:
### What happened

This is similar to #454, but it works on an iPhone, but when you access it on a MacOS, regardless of the browser used, it will result in an infinite loop of handshake and disconnection.

You can see in the logs the iPhone is working correctly while the MacOS web is dropping the connections without any reason. 

[image]

### Client

Mainsail

### Browser

Safari

### How to reproduce

1. Follow RatRig's installation instructions https://os.ratrig.com/docs/introduction/
1. After the setup, open `http://ratos.local` or `http://<IP_ADDRESS>`.
1. Open the networks tab, and see the WebSocket infinite reconnection request

Only port 80 does this, but when you visit port `7125`, or `ratos.local/configure` it works perfectly. 

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/12897270/moonraker.log)

Arksine, 2023-10-15:
Judging by the explanation and the log entries, this appears to be an issue with the reverse proxy.  As you indicate, the websocket works fine when connecting direct to Moonraker, but drops when connecting to port 80 through the proxy.  There are configuration and log files that should be available to help troubleshoot this depending on the proxy being used.

It appears you are using RatOS, so I would recommend contacting RatRig for assistance here.  I suspect they use nginx, but I do not know for sure, nor do I know how they have it configured.

SpaghettiC0des, 2023-10-16:
@Arksine, thanks for the fantastic work and fast response!

I already raised this issue to the RatOS repo.

Additional info: I last used my printer about a month ago. I could still connect before I upgraded all components to the latest version, so it could be a Moonraker or Mainsail update that broke it?

Arksine, 2023-10-16:
If the system packages were updated then its possible that the reverse proxy was part of that.  If we are dealing with nginx then I think a broken update would be widespread, I'm sure there are plenty of Mainsail users that connect over OS X.

The other possibility is that the front end may be the source of the disconnection.  I'm not sure what the RatOS configuration utility does, but if it can successfully connect to the websocket over port 80 then Mainsail should be able to as well.  It appears that Mainsail just released 2.8.0 last week, so you may want to try reverting to 2.7.1.

SpaghettiC0des, 2023-10-17:
Hmm, interesting. It works when I use the hosted version of mainsail. 

[image]

SpaghettiC0des, 2023-10-17:
This guy has an exact issue as me https://www.reddit.com/r/klippers/comments/16u2jt9/cant_connect_to_moonraker_ip_on_mac_but_can_on/?utm_source=share&utm_medium=web2x&context=3

Arksine, 2023-10-17:
> Hmm, interesting. It works when I use the hosted version of mainsail.

That seems to be the expected outcome, I don't believe the remote hosted version of Mainsail will attempt to connect over port 80.

> This guy has an exact issue as me

The timing suggests that it isn't a regression in Mainsail 2.8.0, however its possible that an update to MacOS introduced some kind of compatibility issue.  It might be useful to see if Fluidd works.

Additionally it could be helpful to check the reverse proxy logs.  It looks like RatOS is a fork of MainsailOS, which depends on `nginx` as the reverse proxy.  There should be two log files available for download through Mainsail, `mainsail-access.log` and `mainsail-error.log`.  These are actually nginx log files.  If you can upload them they may provide some useful info.

SpaghettiC0des, 2023-10-18:
@Arksine , I checked it, and unfortunately, it is empty.

SpaghettiC0des, 2023-10-18:
I confirmed this is indeed a MacOS issue; I tried it on an Intel Mac with a Monterey OS, and it works. 

I am still looking for what is causing this in the new MacOS version.

Arksine, 2023-10-18:
Its possible that Mainsail has a issue running in the browser on that specific version of MacOS.  I would try installing Fluidd on the printer host and see if loads successfully.   To install fluidd you can SSH into your machine and run something like the following:
```
cd ~
mv mainsail mainsail-bak
mkdir fluidd
cd fluidd
wget https://github.com/fluidd-core/fluidd/releases/download/v1.26.0/fluidd.zip
unzip fluidd.zip && rm fluidd.zip
ln -s ~/fluidd ~/mainsail
```

The above unzips fluidd into its own directory, then creates a symbolic link named `mainsail` to it.  This will direct `nginx` to load Fluidd rather than Mainsail.   After this is complete you should be able to navigate to the printer using your browser and observe what happens.

When you want to revert back to Mainsail:
```
cd ~
rm mainsail
mv mainsail-bak mainsail
```

SpaghettiC0des, 2023-10-18:
@Arksine thanks! This one works, so it's a mainsail issue then?

Arksine, 2023-10-18:
Yes, if Fluidd works and successfully connects to the websocket on port 80 then it sounds like an issue in Mainsail

Arksine, 2023-10-27:
I'm going to go ahead and close this, hopefully the issue has been (or will be) resolved upstream.

closed 2023-10-27 by Arksine

meteyou, 2023-10-30:
> Yes, if Fluidd works and successfully connects to the websocket on port 80 then it sounds like an issue in Mainsail

fluidd dont use the reverse proxy (port 80). its like the hosted mainsail version. you have to enter your connection and per default, it connects to port 7125.

Arksine, 2023-10-30:
Ah, I didn't realize that.  It sounds like it could be back to an issue with nginx or its configuration.