Arksine/moonraker #727: Generic HTTP type does not seem to work

benjhess, 2023-10-15:
### What happened

I configured a power section for my printer and configured it with the generic http type:

```
[power printer]
type: http
on_url: 
  http://192.168.50.67/apps/api/59/devices/88/open?access_token=some-token
off_url: 
  http://192.168.50.67/apps/api/59/devices/88/close?access_token=some-token
status_url: 
  http://192.168.50.67/apps/api/59/devices/88?access_token=some-token
response_template:
  {% set resp = http_request.last_response().json() %}
  {resp["attributes"][1]["currentValue"]}
initial_state: off
bound_services: klipper
```

I use a tapo p100 switch, accessible via a simple http api on my hubitat hub. I tested the on/off/status urls via curl on the rpi where klipper runs, and it works as expected. e.g:
```
# response format and http method GET is the same for on/off/status
$ curl -s http://192.168.50.67/apps/api/59/devices/88 | jq '.attributes[1]["currentValue"]'
"off"
```

However, the printer is always shown as turned on in mainsail menu, even if the api returns an off status. Therefore, clicking the printer button in mainsail can only issue a printer off action, which does not really execute an off request to the api. Moonraker constantly pulls the status every few seconds.

Mainsail logs when pressing the printer switch:
```
2023-10-15 15:38:37,364 [common.py:_log_request()] - Websocket Received::{"jsonrpc": "2.0", "method": "machine.device_power.off", "params": {"printer": null}, "id": 5375}
2023-10-15 15:38:37,364 [common.py:_log_response()] - Websocket Response::{"jsonrpc": "2.0", "result": {"printer": "init"}, "id": 5375}
```

moonraker version: v0.8.0-186-g2641fc54
mainsail version: v2.8.0

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Use a simple http request api (nodejs example):

```
const http = require('http');
const url = require('url');

let currentState = 'off';

const server = http.createServer((req, res) => {
    const parsedUrl = url.parse(req.url, true);
    res.setHeader('Content-Type', 'application/json');

    switch (parsedUrl.pathname) {
        case '/on':
            currentState = 'on';
            res.end(JSON.stringify({ state: currentState }));
            break;

        case '/off':
            currentState = 'off';
            res.end(JSON.stringify({ state: currentState }));
            break;

        case '/status':
            res.end(JSON.stringify({ state: currentState }));
            break;

        default:
            res.statusCode = 404;
            res.end(JSON.stringify({ error: 'Not found' }));
            break;
    }
});

const PORT = 3000;
const HOST = '0.0.0.0'; // Bind to all available network interfaces
server.listen(PORT, HOST, () => {
    console.log(`Server is running on http://${HOST}:${PORT}`);
});
```

moonraker config:
```
[power printer]
type: http
on_url: 
  http://192.168.50.2:3333/on
off_url: 
  http://192.168.50.2:3333/off
status_url: 
  http://192.168.50.2:3333/status
response_template:
  {% set resp = http_request.last_response().json() %}
  {resp["status"]}
initial_state: off
bound_services: klipper
```

Moonraker will pull status every few seconds. Printer is shown as on in mainsail but should be off due to status http response. Clicking the button will not trigger an off http request to the example api...

Arksine, 2023-10-15:
You will need to attach the full log.  Judging by the snippet the device isn't passing initialization.

benjhess, 2023-10-16:
I have just truncated the logs and restarted the pi, to get clean logs of the problem... and it magically started to work. I have no idea why. Config is still the same 🫠🙈

closed 2023-10-16 by benjhess