Arksine/moonraker #729: [feature request] record speed and flow history

tippfehlr, 2023-10-20:
### Is your feature request related to a problem? Please describe

Not really, but this could be helpful in diagnosing the actual speeds and flow rates of the printer.

### Describe the solution you'd like

Store the values for speed and flow, so a frontend can access them:

Originally I asked for a speed and flow graph in mainsail (https://github.com/mainsail-crew/mainsail/issues/1616#issue-1953928986)

tippfehlr, 2023-11-11:
does this sound interesting? If yes, maybe I’ll take the time and implement this.

joachimth, 2023-11-16:
Absolutely, this sounds interesting. When troubleshooting while printing, it will be nice too have this added graphs.