Arksine/moonraker #731: Spoolman integration does not wort

Mr-Boshi, 2023-10-24:
### What happened

Hi!
I've deployed the Spoolman with postgress for db in docker as a stack. It runs ok, the webui works pretty good but I cannot select any spool in moonraker. The logs show that no matter what spool I select or no matter the ID I pass to the set_active_spool macros it sends the request with the ID=0:
`"PUT /spool/0/use HTTP/1.1" 404 Not Found`

I've opened an [issue at the Spoolman repo](https://github.com/Donkie/Spoolman/issues/171) and the author responded that this is a Moonraker problem. So here I am...

I am running the latest Moonraker `v0.8.0-186-g2641fc54`

[Here is a video](https://www.reddit.com/r/klippers/comments/17fkr0q/cant_select_spool_in_spoolman/) of how it looks in mainsale I've posten on Reddit.

### Client

Mainsail

### Browser

Chrome, Microsoft Edge

### How to reproduce

Deploy a Spoolman with postgress, add a spool and try to select it in the webui or with a macro.

This is the contents of the compose file:

```
spoolman:
    image: ghcr.io/donkie/spoolman:latest
    restart: unless-stopped
    container_name: spoolman
    ports:
      - "7912:8000"
    environment:
      - TZ=Europe/Moscow
      - SPOOLMAN_DB_TYPE=postgres
      - SPOOLMAN_DB_HOST=spoolman-db
      - SPOOLMAN_DB_NAME=spoolman
      - SPOOLMAN_DB_USERNAME=app
      - SPOOLMAN_DB_PASSWORD=app
      - SPOOLMAN_DB_PORT=5432

  spoolman-db:
    image: docker.io/library/postgres:14
    restart: unless-stopped
    container_name: spoolman-db
    volumes:
      - ./spoolman/db/data:/var/lib/postgresql/data
      - ./spoolman/db/config/sql:/config/sql
    environment:
      POSTGRES_DB: spoolman
      POSTGRES_USER: app
      POSTGRES_PASSWORD: app
    healthcheck:
      test: ["CMD-SHELL", "pg_isready -U $$POSTGRES_USER -d $$POSTGRES_DB"]
```

Arksine, 2023-10-24:
It is necessary to attach `moonraker.log` showing a reproduction of the issue.  It may be helpful to provide `klippy.log` as well if there is an issue changing the spool via macro.

Mr-Boshi, 2023-10-25:
Thank you for replying, I was about to post moonraker.log but I have noticed that the spool selection works correctly in Klipperskreen. So it must be not a moonraker bug but a mainsail's instead.

Arksine, 2023-10-25:
Ok, thanks.

closed 2023-10-25 by Arksine

rufinus, 2023-10-29:
For documentation - and if others stumble upon this:

Had this once after a moonraker update - mainsail allows for setting but the selected spool is not shown - in the spoolman logs you see requests for spool_id=0.

The fix is rather easy - restart moonraker service and hard refresh mainsail (to ensure fresh connections). Selecting should work now.