Arksine/moonraker #734: spoolman causes json-rpc request error

killboxalpha, 2023-10-27:
### What happened

installed spoolman upon restart of the moonraker service printer boots and is usable from the screen, but unavailable on mainsail/moonraker. can only get to initialization screen with the following errors in mookraker.log

```
2023-10-26 20:30:34,490 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.spoolman.proxy, Code: 500, Message: HTTP Request Error: http://192.168.4.16:7912/api/v1/info
2023-10-26 20:30:34,491 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.spoolman.proxy, Code: 500, Message: HTTP Request Error: http://192.168.4.16:7912/api/v1/health
2023-10-26 20:30:34,491 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.spoolman.proxy, Code: 500, Message: HTTP Request Error: http://192.168.4.16:7912/api/v1/vendor
2023-10-26 20:30:35,287 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: server.spoolman.proxy, Code: 500, Message: HTTP Request Error: http://192.168.4.16:7912/api/v1/spool/-1
```

if I restart the service multiple times, it will eventually load. 

Removal of the spoolman integration is clear the problem as well

### Client

Mainsail

### Browser

Chrome, Firefox, Microsoft Edge

### How to reproduce

standalone install of spoolman, with basic moonraker integration. 

upon restart of the moonraker service with the spoolman config the front end of mainsail will fail to load. 

`[spoolman]
server = http://192.168.4.16:7912
sync_rate = 60
`

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/13184476/moonraker.log)

Arksine, 2023-10-29:
It appears that Moonraker cannot reach your spoolman instance.  That said, I don't think this is something that should block initialization in Mainsail.  It is something that likely needs to be addressed there.

killboxalpha, 2023-10-29:
given this in directly in the moonraker config I would assume it has something to do with it. 

while the spoolman config for moonraker is active, i can not get to either the front end of mainsail, or the spoolman page, restarting the moonraker service allows both the mainsail and spoolman pages to be reached. once the integration is commented out of the moonraker.conf both services will work, but are not linked.

Arksine, 2023-10-29:
Adding the `[spoolman]` to the configuration loads the spoolman component.  This enables the filament tracking logic used to update Spoolman's database.  In addition, it allows Moonraker to act as a proxy between a front end (Mainsail in this case) and Spoolman.  When the front end makes a proxy request it must properly handle the response, even if the response is an error.

When you remove `[spoolman]` from the configuration, the component is disabled.  Mainsail likely detect this and doesn't attempt to use Moonraker as a proxy to communicate with spoolman...thus no errors.

closed 2023-11-01 by killboxalpha