Arksine/moonraker #735: Button on_press & on_release apprear to be reveresed

gandlers, 2023-10-29:
### What happened

Button on_press actions happens when button released & on_release action happens when the button is pressed.

Discovered when trying to follow the Reset button example from the moonrakers docs

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Configured 2 relays and 2 push buttons to toggle them. Buttons configured with pullups and connect to ground when pressed.
Intend to use a button to power printer on and off, so need to set it to require pressing for 2 seconds to ensure it's not pressed by accident mid print so need to use on_release and event.elapsed_time. However couldnt get the button reset example from the moonraker docs working, The on_release action would happen as soon as button pressed.

excerpt from moonraker.conf
```
[power printer]
type: gpio
pin: gpio17
off_when_shutdown: True
initial_state: off
#on_when_job_queued: True
#locked_while_printing: True
#restart_klipper_when_powered: True
#restart_delay: 1.

[power gantry_light]
type: gpio
pin: gpio27
off_when_shutdown: True
initial_state: off

# Button 2
[button btn_gantry_light]
type: gpio
pin: ^gpio23
minimum_event_time = 0.2
on_press:
  {% do call_method("machine.device_power.post_device", device="gantry_light", action="toggle") %}

# Button 4
[button btn_TEST]
type: gpio
pin: ^gpio24
#minimum_event_time = 0.2
on_press:
  {% do send_notification("Button 4 pressed") %}
  {% do call_method("machine.device_power.post_device", device="printer", action="toggle") %}
on_release:
  {% do send_notification("Button 4 released") %}
  {% do call_method("machine.device_power.post_device", device="gantry_light", action="toggle") %}
```

Pressing button 2 only toggles the gantry_light  when the button is release and not when pressed as configured
pressing button 4 toggles the gantry light when pressed and toggles the printer relay when relased, despite being configured to operate in the reverse.

### Additional information

Initially though it was that the button inputs needed inverting eg !^gpio24, but this resulted in the button not being detected.
Hopefully this is actually user error as I've only recently migrated from marlin to mainsail/klipper.

Arksine, 2023-10-29:
The button does need to be inverted, however the prefix in `!^gpio24` is ordered incorrectly.  It should be `^!gpio24`.

gandlers, 2023-10-30:
thanks for your response.
I'm a little embarrased now. I can see now that the docs learly show the ^ before the !.  Not sure how I missed that! 
This is obviously why my attempt to invert the input failed. 
Unfortunately I'm not home until Wed/Thur to be able to test this but I'm sure that this is the issue as moornraker thinks the button is released when I press it and vice versa.
Hopefully I can confirm its was user error later in the week and close this.

gandlers, 2023-11-04:
I can confirm this was user error on my part
Using the correct syntax for the inverted gpio input (^!gpio24) is works ans expected with on_release triggering correctly when the button is released. 
Sorry for the false flag

closed 2023-11-04 by gandlers