Arksine/moonraker #744: [power printer] does not shutdown klipper/bound services on raspberry pi boot

Snelso91, 2023-11-11:
### What happened

Following the docs from https://moonraker.readthedocs.io/en/latest/configuration/ and https://docs.mainsail.xyz/overview/quicktips/printer-power-switch, I have setup a `[power]` section in my moonraker.conf, which controls a mosfet switch via gpio which cuts power to the printer mainboard:
```
[power printer]
type: gpio
pin: gpio23
initial_state: off
off_when_shutdown: true
on_when_job_queued: true
restart_klipper_when_powered: true
bound_services:
 klipper
 klipper-mcu
```

Turning on the associated `printer` switch in the Mainsail Power Devices menu will start the bound services and also restart klipper when it is ready due to `restart_klipper_when_powered: true`. Similarly, turning off the `printer` switch will stop the bound services (klipper and klipper-mcu), which also triggers Mainsail to show a "Printer is powered off" section.
[image]

However, if you boot/reboot the raspberry pi that is running all of this software, the initial state will be a klipper error screen instead of the "Printer is powered off" screen. On inspection you can see that this is because the bound services (klipper and klipper-mcu) are actually running and have not been stopped by moonraker, despite the `printer` switch clearly being in the off position due to its `initial_state: off`.
[image]

### Client

Mainsail

### Browser

Chrome

### How to reproduce

1. Reboot/fresh boot raspberry pi, with moonraker configured with a `[power]` section with `initial_state: off` and `bound services: klipper`.
2. Observe that moonraker does not shutdown klipper even though the `initial state` is off.

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/13325781/moonraker.log)

Arksine, 2023-11-11:
This is expected behavior.  When the device is already off the initial state is satisfied.  Bound services are only triggered when state changes.

Snelso91, 2023-11-11:
@Arksine ok, in that case can I file this as a feature request instead?
This is a useful as a quality of life feature because it removes the error I showed in the second screenshot, stops klipper from trying to reconnect to the mcu and filling the logs with messages about the mcu being disconnected, and gives you the more polished "printer is off" screen at boot.

As an example of a use case, when I first boot the printer I might leave it in this "standby" state for a long time with the mainboard powered off and just the pi running. Then whenever I upload a job to the pi, it will automatically switch the mainboard on, begin the print, and the switch the mainboard off again when the printer has finished and cooled. In this way you can have a sort of "always on" printer whilst only having a few watts of power draw from just the pi and its peripherals in the "standby" mode.

Arksine, 2023-11-15:
I went ahead and added this in commit 6c6b7dcff0837a0946de6f7e39ceb8772a34d4cf.  That said, there is still the possibility that it won't work as you hope.  If during boot systemd loads Klipper after Moonraker, it may start after Moonraker has already initialized your power device.  In this scenario the service will be running.

closed 2023-11-15 by Arksine

Snelso91, 2023-11-16:
>If during boot systemd loads Klipper after Moonraker, it may start after Moonraker has already initialized your power device. In this scenario the service will be running.

Thanks, I just updated and tried it and it works perfectly. I tested on a fresh boot twice and both times klipper was shutdown by the time I managed to reconnect to Mainsail, so I was greeted with the "printer is powered off" screen both times. So it seems the edge case you mentioned is thankfully not occurring on my raspberry pi 3A+.

Arksine, 2023-11-16:
Its probably rare.  The best fix for your use case would be to disable the klipper service so that its only started manually.  Unfortunately Moonraker cannot currently detect disabled systemd services and the modifications required to do so are a bit more involved.