Arksine/moonraker #746: API doesn't show cameras defined in conf, until a there is a camera defined from UI

vajonam, 2023-11-16:
### What happened

I have added a couple of cameras to my .conf file (webcam foo), they don't show up using the 

`GET /server/database/list`

until I go into one of the UI's and add webcam, then both the UI defined and .conf defined webcams' show up. 

### Client

Mainsail, Fluidd

### Browser

Chrome

### How to reproduce

Clean moonraker, add a webcam in .conf

`GET server/database/item?namespace=webcams` returns an empty list. 
`GET /server/database/list`

Goto fluidd or mainsail, add new camera, then `GET server/database/item?namespace=webcams`  returns both .conf defined webcams and UI defined webcams.

Arksine, 2023-11-16:
Webcams added to the configuration are not added to the database.  The [webcam APIs](https://moonraker.readthedocs.io/en/latest/web_api/#webcam-apis) must be use to access them.

vajonam, 2023-11-16:
Hmm.. understood. Possible a bug somewhere else in the UIs. behavior is the UI's dont show .conf webcams until one is added to the db then it shows it.. I will narrow this down to see what is causing it.

closed 2023-12-14 by vajonam