Arksine/moonraker #747: HappyHare v2 (ERCF) pause does not trigger notifier

Cyberwizzard, 2023-11-16:
### What happened

When using HappyHare v2 for my ERCF, it calls "PAUSE" when there is an issue with the print like a clog during a color swap.

I have set up the notifier component which works for all normal events, like print start, stop or manual pause. I get PushOver events for all of these just fine.

However, when using HappyHare v2, the call to PAUSE does not seem to trigger the event notifier.

Is there something I can do to troubleshoot this? Alternatively, can I trigger the notifier in Moonraker from a macro somehow? 

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Install a Voron, install HappyHare v2 on your ERCF, set up notifier for Moonraker like normal - HHv2 induced pauses do not trigger an event.

Arksine, 2023-11-16:
It appears that HappyHare v2 is some kind of 3rd party extension for Klipper.  Unfortunately I can't provide much help here, I do not know what the module has changed with regard to pausing. I would suggest asking the author of the module.

It is possible to [emit notifications from macros](https://moonraker.readthedocs.io/en/latest/configuration/#notifying-from-klipper).

Arksine, 2024-02-04:
As no one has been able to provide further feedback I'm going to close this.  If you are still having issues I recommend asking for assistance on the Klipper Discourse.

closed 2024-02-04 by Arksine