Arksine/moonraker #748: Fallback object height parsing from PrusaSlicer incorrect for heights <1mm

Jabele10, 2023-11-16:
### What happened

While trying to find a problem of the object height not getting parsed I inspected the code used for this parsing and noticed something interesting in _[moonraker/components/file_manager/metadata.py](https://github.com/Arksine/moonraker/tree/master/moonraker/components/file_manager/metadata.py)_ around line 362:
```
def parse_object_height(self) -> Optional[float]:
    matches = re.findall(
        r";BEFORE_LAYER_CHANGE\n(?:.*\n)?;(\d+\.?\d*)", self.footer_data)
    if matches:
        try:
            matches = [float(m) for m in matches]
        except Exception:
            pass
        else:
            return max(matches)
    return self._parse_max_float(r"G1\sZ\d+\.\d*\sF", self.footer_data)
```
The fallback uses the regex `G1\sZ\d+\.\d*\sF` which will work in most cases.
The problem is that PrusaSlicer produces G-Code without a leading zero. The G-Code `G1 Z.4 F9000` as output of PrusaSlicer won't be found by that regex since it requires to have a digit before the dot.
As I see the definition of `parse_max_float`, the regex `G1\sZ\d*\.\d*\sF` (so it doesn't require the digit) won't just work since this function requires a digit as well (don't know if you could just change that too).

It's not really a big deal since this is just fallback, however in some cases it could lead to unpredictable behaviour.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

You would get suspicious behaviour under this circumstances:
- Use PrusaSlicer (version shouldn't matter, however I'm using 2.6.1)
- Prevent the default value from working (for example by putting something in between `;BEFORE_LAYER_CHANGE` and `;{layer_z}` at the Layer Change G-Code in PrusaSlicer)
- Sending a file with a height < 1 mm to moonraker

### Additional information

I suppose the log files don't really matter since I actually linked the code in question.
For client and browser these are just the ones I'm using but since the problem is in the code for the upload, these won't matter at all.

Arksine, 2023-11-17:
Interesting.  Older versions of PrusaSlicer always included a leading 0, this must have changed at some point.  Regardless, I agree that this is something that should be corrected.

Arksine, 2023-11-20:
This should be resolved by commit be23f206ee829f7e780973098fa6235b56e3d5c0.  Honestly the float pattern throughout the entire module could be improved so I did a bigger refactor.

closed 2023-11-20 by Arksine