Arksine/moonraker #749: Moonraker exclude_object processor is outputting wrong coordinates in the EXCLUDE_OBJECT_DEFINE section

Snelso91, 2023-11-17:
### What happened

If I take a gcode file from superslicer which has the objects labelled at the start of the file:
```
; object:{"name":"test-pieces_cone-10mm-tall","id":"test-pieces_cone-10mm-tall.stl id:0 copy 0","object_center":[117.000000,117.000000,0.000000],"boundingbox_center":[117.000000,117.000000,10.000000],"boundingbox_size":[10.000000,9.927090,20.000000]}
; plater:{"center":[117.000000,117.000000,0.000000],"boundingbox_center":[117.000000,117.000000,10.000000],"boundingbox_size":[10.000000,9.927090,20.000000]}
```
Then after it is uploaded to mainsail/moonraker, the resulting coordinates outputted from moonraker by its processor are similar to the original, but not the same:

```
; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 1 known objects
EXCLUDE_OBJECT_DEFINE NAME=test_pieces_cone_10mm_tall_stl_id_0_copy_0 CENTER=117.127,116.992 POLYGON=[[112.304,112.339],[112.304,121.661],[121.702,121.661],[121.702,112.339]]
```

From looking closely at the polygon coordinates, it seems the numbers are about 0.3mm closer to the true centre point (117,117) than they should be, but I no idea why you would want to change the numbers like this?

Also, the strangest part is not only that the 4 corners of the XY bounding box are wrong, but that the center point coordinates are not even the same as the original given by the slicer, which could have been simply copied (`CENTER=117.127,116.992` instead of the original `117,117`).

I understand that computers have a limit for precision for floating point numbers, but this seems way too excessive for that, even a simple calculator would not have an error this large when subtracting/adding numbers from a center point.
This means that currently in other macros that use this exclude_object definition such as adaptive bed meshing, I need to manually expand the incorrect `EXCLUDE_OBJECT_DEFINE` bounding box again by 0.5mm or so to account for this error.

### Client

Mainsail

### Browser

Chrome

### How to reproduce

1. Upload the attached original gcode to Mainsail.
2. After it has been processed, the file will now have incorrect EXCLUDE_OBJECT_DEFINE coordinates.

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/13393980/moonraker.log)

Original gcode file before upload to mainsail (change extension back to .gcode):
[test-pieces_cone-10mm-tall original.txt](https://github.com/Arksine/moonraker/files/13394007/test-pieces_cone-10mm-tall.original.txt)

Gcode file after upload to mainsail and processing (change extension back to .gcode):
[test-pieces_cone-10mm-tall processed.txt](https://github.com/Arksine/moonraker/files/13394016/test-pieces_cone-10mm-tall.processed.txt)

Arksine, 2023-11-17:
Moonraker itself doesn't perform the object processing, it depends on the [preprocess_cancellation](https://github.com/kageurufu/preprocess_cancellation) library for this functionality.

Snelso91, 2023-11-17:
@Arksine ok, so I should report this bug on their github instead?

Arksine, 2023-11-17:
If you believe it is a bug, then yes.  The way it calculates the bounding box may differ from how PrusaSlicer calculates it.  I don't believe it uses the comments, since they may not be there.  It calculates its own bounding box based on processing the gcode file.  It seems to arrive a different conclusion than PrusaSlicer.

FWIW, PrusaSlicer 2.7 will support Klipper objects natively, which is nice.

Arksine, 2023-11-17:
Looking closer, I doubt this is a bug in `preprocess_cancellation`.  If you create a bug report on `preprocess_cancellation` you are likely going to need to show that a processed object did not fully cancel during normal operation.  Calculating boundaries that aren't a perfect match for PrusaSlicer's calculations would not make it a bug.

Snelso91, 2023-11-17:
@Arksine ok, after I looking closer I think I have understood the discrepancy a bit better.

First of all, in superslicer, I have a number of compensations set up to pre-correct the g-code so that the finished printed object is dimensionally accurate to the original model/stl. One of these compensations is XY first layer compensation (aka elephants foot compensation) which is set to -0.25mm (per side/radius).
This correction fades out to 0 over the first 10 layers, so it means that for example the X minimum in the gcode is on layer 7, which is what is reported by preprocess_cancellation. This is due to a complex interaction between the faded XY compensation (shrink some amount between 0 and 0.25), a material shrinkage compensation (causing an expansion of the gcode by 0.3%), and the fact that the gcode does not account for half the width of extruded filament line (shrink 0.2mm).

Secondly, the bounding box reported by superslicer is actually the bounding box of the stl model before slicer corrections. I can test this for example by setting material shrinkage to 50%, which will expand the gcode in XY by 2X (which can be clearly seen in the gcode preview), and yet the bounding box remains the same.
So in that case it is clear that it would be incorrect to use the superslicer bounding box, since the printhead would actually be moving far beyond those bounds.

So in the end, although the bounding box of the physical finished printed part will be the same as what superslicer reports for the stl (X112,Y112.036) to (X122,Y121.964), it is necessary/correct to use the gcode bounding box for the purposes of bed mesh probing etc.
In this specific case, although the gcode box is slightly smaller than the space the physical part will occupy on the bed after printing, this shouldn't be an issue so long as you space parts apart enough to allow for the expansion caused by elephants foot for example.

closed 2023-11-17 by Snelso91

Arksine, 2023-11-17:
I see.  I would point out that a fraction of a millimeter is not going to impact the mesh.  Moves outside of the mesh have their adjustment clamped to the nearest point a the edge of the mesh.  The change in adjustment by probing a half millimeter out would be miniscule.