Arksine/moonraker #754: Using a oneshot token to login a user without password (and get a jwt token)

basz, 2023-11-24:
### Is your feature request related to a problem? Please describe

I need to hand out a jwt for user to connect on a dashboard to the api. The backend has access via the apiKey. If I'm not mistaking only a login (with password) or a create user (also with password) will return a jwt. Is there a way around this? In other word can a token be generated without actually becoming that user?

### Describe the solution you'd like

The obvious one would be to return a jwt when a user logs in without a password but with a valid oneshot token

Arksine, 2023-11-24:
It is not possible to login without a password.  A oneshot token has a very specific purpose; to allow authentication for http requests where it is not possible to modify request headers, such JS Websockets.  The token's credentials are derived from the original request, using the API Key to request a token would not be sufficient to authenticate as another user.

My suggestion would be to have your backend proxy the login request rather than use the API Key.

basz, 2023-11-24:
Thank you for responding...

Proxying the login only wouldn't work as our users aren't supposed to login into moonraker. They are logging into our dashboard via seperate credentials and ideally would be able to then see their printers.

Expanding on your suggestion. Perhaps it would be possible to proxy *all* websockets request through the backend and use the api key to authenticate each session. But I would rather not do that.

Deleting and recreating a user with a random password would work, but deleting an existing user would probably invalidate existing jwt's. So different (dashboard) user would interfere...

My backend could also keep track of JWT created for a single user and distrubute that to all clients. But that also seems cumbersome.

There is no way to exchange an api key for a jwt for a (virtual) user?

Arksine, 2023-11-27:
>  Proxying the login only wouldn't work as our users aren't supposed to login into moonraker. They are logging into our dashboard via seperate credentials and ideally would be able to then see their printers.

Ok.  If the intent is to connect the frontend directly to Moonraker it may be best to have it authenticate directly.  Presumably users would already have that configured via `trusted_clients` or through a user they have created.

> Expanding on your suggestion. Perhaps it would be possible to proxy all websockets request through the backend and use the api key to authenticate each session. But I would rather not do that.

That is an option and probably the best one if your dashboard is hosted outside of the wan.   If its intended to run on the LAN then it would likely be fine to have the frontend request the API Key from your backend, then use it to authenticate the websocket connection.   No need to save the API Key to local storage, just request it when needed.

> My backend could also keep track of JWT created for a single user and distrubute that to all clients. But that also seems cumbersome.

If you absolutely want to use tokens this is the best option.   Creating a user and generating tokens is relatively easy.  When you create or log in a user you receive an access token and a refresh token.  The refresh token can be used to create as many access tokens as you need (via `/access/refresh_jwt`).  The cumbersome part of JWT authentication is actually using them:

- Access tokens expire 60 minutes after being issued.  The refresh token's expiration is configured in `moonraker.conf`, the default is 90 days.

- Before using a token the frontend should decode it to verify that hasn't expired.  If the access token has expired the refresh token to can be used to request a new one.  If the refresh token has expired, a new login is necessary.

- Logging out (via `/access/logout`) invalidates all outstanding tokens, so your frontend must handle 401 unauthorized conditions even when it thinks a token is valid.

> There is no way to exchange an api key for a jwt for a (virtual) user?

No, a real user is required to generate a JWT.