Arksine/moonraker #757: FR: Load file metadata on "Print", so they can be evaluated in macro's

ccatlett1984, 2023-11-27:
### Is your feature request related to a problem? Please describe

Moonraker already contains all the metadata fields for a loaded gcode file, I would like to have them loaded as variables when a file is started for printing, so they are dynamically available to use in macros executed for that print job. 

### Describe the solution you'd like

When a print is started, load the metadata fields for that file as variables, prior to evaluation of any "start_gcode" from the slicer, this allows anything defined by the slicer to over-ride the metadata, but makes all the fields available for use.

### Describe alternatives you've considered

re-slicing all files, to add an additional variable into the start_gcode in the slicer, each time a variable needs to be added, when the fields are already present in the file metadata just seems like a horrible time sink. Doing this in an automated/dynamic fashion in moonraker seems a much more tenable solution.

Arksine, 2023-11-27:
I think what you are looking for is a feature addition to Klipper.  Currently there is no way for Moonraker to send this data to Klipper, nor is there a way for Klipper to request it from Moonraker.

I would suggest starting a thread on the [Klipper Discourse](https://klipper.discourse.group/) to see if you get any traction.

ccatlett1984, 2023-11-27:
Linking Klipper Discourse thread for visibility:
https://klipper.discourse.group/t/fr-on-print-start-pull-metadata-from-moonraker-so-it-is-available-for-macro-variables/11908