Arksine/moonraker #759: cancellation_preprocessing not ( well ) working on idamaker

Pigi-102, 2023-11-28:
### What happened

the routine place EXCLUDE_OBJECTS after START_PRINT in gcode, thus KAMP cannot work as expected.
I've also added an M117 as suggested elsewhere with no joy.
Version in use:
fluidd v1.26.3
klipper v0.12.0-10-gea2f6bc0
Klipper-Adaptive-Meshing-Purging v1.1.2-7-g3eca87b7
mainsail v2.8.0
moonraker v0.8.0-204-gfb15b2a3

( cancellation_preprocessing v0.2.0 )

### Client

Mainsail, Fluidd

### Browser

Chrome, Firefox

### How to reproduce

Generate a file from ideamaker and upload. EXCLUDE_OBJECTS would be in the wrong place ( after START_PRINT )

### Additional information

( partial ) gcode from ideaMaker 
`
Sliced by ideaMaker 4.3.3.6560, 2023-11-28 20:29:55 UTC+0100
;Dimension: 255.000 255.000 260.000 0.400
;Plate Shape: 0
;Origin Center: 0
;Extruder Offset #1: 25.000 0.000
;Filament Diameter #1: 1.750
;Filament Diameter #2: 1.750
;Filament Compensation #1: 95.00
;Filament Compensation #2: 90.00
;Filament Density #1: 1240.00
;Filament Density #2: 1240.00
;Filament Type #1: PLA
;Filament Type #2:
;Filament Name #1: [Raise3D] PLA 1.75mm
;Filament Name #2:
;Firmware: Marlin
;Bounding Box: 92.787 162.227 97.759 167.240 0.000 10.000
;Model Gap: 0.000
M221 T0 S95.00
M117
#Start print
START_PRINT BED_TEMP=85 EXTRUDER_TEMP=220
M1001
;TOTAL_NUM: 4
;PRINTING_TIME: 0
;REMAINING_TIME: 1806
;LAYER:0
;Z:0.200
;HEIGHT:0.200
M106 S0
;PRINTING: NON-OBJECT
;PRINTING_ID: -1
G1 F3000 E-5.5000
G0 F12000 X99.560 Y103.105
G0 F600 Z0.200

`
result after upload:

`
;Sliced by ideaMaker 4.3.3.6560, 2023-11-28 20:37:10 UTC+0100
;Dimension: 255.000 255.000 260.000 0.400
;Plate Shape: 0
;Origin Center: 0
;Extruder Offset #1: 25.000 0.000
;Filament Diameter #1: 1.750
;Filament Diameter #2: 1.750
;Filament Compensation #1: 95.00
;Filament Compensation #2: 90.00
;Filament Density #1: 1240.00
;Filament Density #2: 1240.00
;Filament Type #1: PLA
;Filament Type #2:
;Filament Name #1: [Raise3D] PLA 1.75mm
;Filament Name #2:
;Firmware: Marlin
;Bounding Box: 92.787 162.227 97.759 167.240 0.000 10.000
;Model Gap: 0.000
M221 T0 S95.00
M117
#Start print
START_PRINT BED_TEMP=85 EXTRUDER_TEMP=220
M1001
;TOTAL_NUM: 4

; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 4 known objects
EXCLUDE_OBJECT_DEFINE NAME=Foot3_stl CENTER=113.424,118.543 POLYGON=[[101.742,106.719],[101.742,130.305],[125.305,130.305],[125.305,106.719]]
EXCLUDE_OBJECT_DEFINE NAME=Foot3_3_stl CENTER=113.447,146.446 POLYGON=[[101.742,134.695],[101.742,158.281],[125.305,158.281],[125.305,134.695]]
EXCLUDE_OBJECT_DEFINE NAME=Foot3_2_stl CENTER=141.388,146.457 POLYGON=[[129.694,134.695],[129.694,158.281],[153.257,158.281],[153.257,134.695]]
EXCLUDE_OBJECT_DEFINE NAME=Foot3_1_stl CENTER=141.378,118.543 POLYGON=[[129.694,106.719],[129.694,130.305],[153.257,130.305],[153.257,106.719]]
;PRINTING_TIME: 0
;REMAINING_TIME: 1806
;LAYER:0
;Z:0.200
;HEIGHT:0.200
M106 S0
;PRINTING: NON-OBJECT
;PRINTING_ID: -1

`

Pigi-102, 2023-11-28:
In case you need this is the gcode from slicer:
[Foot3.gcode.txt](https://github.com/Arksine/moonraker/files/13492530/Foot3.gcode.txt)

Arksine, 2023-11-28:
This is an issue with the `preprocess_cancellation` library that performs the processing. 
 It looks similar to https://github.com/kageurufu/preprocess_cancellation/issues/29.

Pigi-102, 2023-11-28:
Sorry @Arksine , can yo help me understand why my issue has been marked as invalid ?
What should I do now ? Should I follow other link ?
Thanks

Pigi-102, 2023-11-28:
Thanks ! It seems that in [https://github.com/kageurufu/preprocess_cancellation/issues/23](url) has already been reported.

Arksine, 2024-02-04:
This should now be resolved with `preprocess-cancellation` version 2.1.

closed 2024-02-04 by Arksine