Arksine/moonraker #760: K1 GCodeMeta Thumbnails width and height

Clon1998, 2023-12-01:
### What happened

A couple of users reported a parsing error in Mobileraker (https://github.com/Clon1998/mobileraker/issues/288).
I tracked it down to the GCodeMeta-Thumbnails data that Moonraker reports. After further investigation, I noticed that for whatever reason, the thumbnail's `width` and `height` fields are `strings` rather than `ints`.
The user that reported this issue is using a K1 and Creality Slicer. Therefore, I manually ran the `metadata.py` to see if I can figure out the reason behind the string types without any luck.

### Client

MobileRaker

### Browser

Other or N/A

### How to reproduce

Slice a file with Creality Print and analyze it.
I could not get the metadata the user provided in the linked issue. Therefore, I do not know if reality implemented something themself.

Arksine, 2023-12-01:
Moonraker doesn't support metadata extraction from Creality Slicer.  Presumably the metadata reported by your users was extracted before they rooted the device and changed to the official repo.

Clon1998, 2023-12-01:
> Moonraker doesn't support metadata extraction from Creality Slicer. Presumably the metadata reported by your users was extracted before they rooted the device and changed to the official repo.

That makes sense and exactly reflects my findings! Thanks for the fast reply.

closed 2023-12-12 by Arksine