Arksine/moonraker #763: Make temp-directory configurable

Plopsi, 2023-12-06:
### Is your feature request related to a problem? Please describe

the Creality K1 (Max) uses a in-ram tempfs /tmp folder with only 100MB (max 200MB is reachable due to memory limit). And when upload larger files the upload fails as moonraker puts the uploads hard into /tmp/moonraker.upload-XXX.mru.

So files larger than 100MB/200MB are currently not uploadable via i.e. mainsail.

### Describe the solution you'd like

Make a configuration option to change the tempfolder and only fallback to the default OS-temp-folder when not set.

### Describe alternatives you've considered

Currently I workaround by uploading larger files not via mainsail (and therefore Moonraker) but with the abused cruality fluidd version called CrealityOS. Works, but would be nicer to have an option to directly use my main UI.

### Additional information

Typical for creality *rolleyes*.

Example of "hardcoded" temp-destination:
https://github.com/Arksine/moonraker/blob/master/moonraker/components/file_manager/file_manager.py#L779C34-L779C34

Arksine, 2023-12-12:
There is no need to add a configuration option to Moonraker to change the temporary directory.  Just add something like the following to `moonraker.env`:
```
TMPDIR="/path/to/custom/dir"
```

fir3bird, 2023-12-26:
@Arksine First: Merry Christmas and thank you very much for your awesome work on moonraker! :-)

Currently I am facing the same issue and tried your advice. Just like Plopsi I am using a K1 from creality.
Unfortunately it still seems files are uploaded to /tmp dir instead of the one I configured in the moonraker.env file.

What am I missing?

[image]

[image]

fir3bird, 2023-12-26:
OK, I now just added `export TMPDIR=/usr/data/printer_data/tmp` to the start() method of the init.d script. And that works fine.
But to be honest it feels a bit hacky.
Maybe this could help you anyway @Plopsi until we find a more cleaner solution?

Arksine, 2023-12-26:
Ah, it appears the linux distro that ships with the K1 uses `sysvinit` rather than `systemd`, so the environment file `moonraker.env` is not used.  I suspect running official Moonraker on that machine will end up being a bit hacky unless Creality opens it up beyond root access.

IIRC the typical way of sourcing varibales in sysvinit is to put them in a "defaults" file then source it in the init script.  For example, in `/etc/default/moonraker`:
```
TMPDIR="/path/to/custom/dir"
```

Then in the init script:
```bash
...
PID_FILE=/var/run/moonraker.pid
DEFAULTS_FILE=/etc/default/moonraker

[ -f $DEFAULTS_FILE ] && . $DEFAULTS_FILE

start() {
...

iandouglas, 2024-03-03:
hey @Arksine thanks for [this comment](https://github.com/Arksine/moonraker/issues/763#issuecomment-1851772275), it fixed exactly what I needed on a Troodon 2.0 Pro system which only gave a 512MB /tmp folder.