Arksine/moonraker #765: Moonraker does not trigger autofs to mount gcodes directory.

trevjonez, 2023-12-12:
### What happened

On a fresh boot moonraker will report a warning about permissions to gcodes also causing mainsail to not show gcode file list GUI options.

```
Moonraker warning
Moonraker does not have permission to access path (/home/pi/nfs/gcodes) for (gcodes).
```

### Client

Mainsail

### Browser

Chrome

### How to reproduce

autofs configuration is fairly simple.

`/etc/auto.master`
```
+dir:/etc/auto.master.d
+auto.master

/home/pi/nfs /etc/gcodes.nfs uid=1000,gid=1000 --timeout=600 --ghost
```

`/etc/gcodes.nfs`
```
gcodes  -fstype=cifs,rw,noserverino,username=somePrinterUser,password=somePassword   ://unRAID/GCodes
```

printer_data/gcodes is symlinked to the mount location ~/nfs/gcodes
[image]

### Additional information

To resolve I typically just run an `M20` gcode to trigger klipper to list the contents in the console thus mounting the directory. From there restarting the moonraker service will turn everything nominal.

Arksine, 2023-12-15:
Thanks.  I just pushed commit 6d8cb762ffca1bd151d98e0cec5b66b966965014 which relaxes the requirements a bit.  It will attempt to list the folder's contents during registration, then use eaccess to check for write permissions.  Registration will continue even if both fail.  That said,  if `inotify` is enabled a warning may still be generated if the autofs service doesn't launch before Moonraker on startup.   If that does occur simply restarting Moonraker should resolve it.

trevjonez, 2023-12-16:
host reboot is usually a sure fire way to trigger the warning and it is no longer with us.

TY for the quick fix sir.

closed 2023-12-16 by trevjonez