Arksine/moonraker #766: Need help, suddenly can't connect to my klipper "Message: Unauthorized"

CptanPanic, 2023-12-14:
### What happened

Literrely, one day it worked, and the next day it didn't . I didn't update anything even. 

### Client

Mainsail

### Browser

Firefox

### How to reproduce

I go to my printer url, and title says "ERROR" , and progress bar just says initializing.  Logs are attached.

[logs.zip](https://github.com/Arksine/moonraker/files/13675168/logs.zip)

Arksine, 2023-12-14:
According to `moonraker.log` you are attempting to connect to Moonraker from an IP that is not trusted (`100.118.222.24`).  Make sure your PC/Phone/etc is connected to the same local network as Moonraker.  It looks liek the IP should be in the range of `192.168.1.x`.

ForceConstant, 2023-12-14:
> According to `moonraker.log` you are attempting to connect to Moonraker from an IP that is not trusted (`100.118.222.24`). Make sure your PC/Phone/etc is connected to the same local network as Moonraker. It looks liek the IP should be in the range of `192.168.1.x`.

Ah thank you. Yes my ip changed in my network. Thanks, that was it.

closed 2023-12-15 by Arksine