Arksine/moonraker #767: file_manager: Klipper configuration file not located in `config` folder?

eyJhb, 2023-12-15:
### What happened

I'm not sure that it makes sense, to emit a warning of `klipper.cfg` is not in the Moonraker config folder? Or rather, I don't see why it is so.

klipper and moonraker are two different types of software, where there is no particular reason, for the configuration to be in the moonraker config directory.
The only reason I can think of, that this is a thing, is because it eases the ability to edit the klipper configuration file from moonraker/gui.

Can someone shed some light on this? 

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Place `klipper.cfg` outside of moonraker config directory.

Arksine, 2023-12-16:
It needs to be in the config folder for Moonraker to edit it.  Moonraker is restricted from writing files outside of the gcode and configuration folders.

While Moonraker and Klipper are different applications, Moonraker is inextricably tied to Klipper.  As such, managing Klipper with Moonraker will require some specific configuration in Klipper to get the most out of it.

closed 2023-12-16 by Arksine

eyJhb, 2023-12-16:
That makes no sense at all. I have no specific reason for editing the config, and no need for it. Please provide a way to disable this warning.

Arksine, 2023-12-16:
Managing Klipper's configuration is part of Moonraker's functionality and the overwhelming number of users do find it useful.

You do not need to move your Klipper configuration to silence the warning, you can create symbolic link to it in the config folder.  I won't make a change to Moonraker to silence the warning, however you are free to fork it and remove the warning if you would like.

eyJhb, 2023-12-16:
Based on what you wrote, the following should silence the warning, right?

```
[moonraker@hephaestus:/var/lib/moonraker/config]$ ls -al
total 16
drwxrwx--- 2 moonraker moonraker 4096 Dec 16 19:53 .
drwxrwx--- 9 moonraker moonraker 4096 Oct 13 19:02 ..
lrwxrwxrwx 1 moonraker moonraker   16 Dec 16 19:53 klipper.cfg -> /etc/klipper.cfg
-rwxrwx--- 1 moonraker moonraker  217 Dec 16 19:58 moonraker-temp.cfg
-rw-rw---- 1 moonraker moonraker  228 Dec 16 19:58 .moonraker-temp.cfg.bkp
```

The warning however persists :)

Arksine, 2023-12-16:
Ah, you are correct and I am mistaken.  Rather than modify the validation procedure I will provide a way to opt out of the validation step.

eyJhb, 2023-12-16:
I would have no issue if the warning was removed, if the configuration file was symlinked into the Moonraker configuration directory. But I'm glad to hear, that it was not just me doing something wrong. :)

A way to opt-out of that specific validation step would also be highly appreciated, but I do _fully_ understand the concerns you present. I can only imagine the amount of support requests Moonraker gets on Discord and other places, that such warnings would be able to aid in fixing.

Can this issue maybe be reopened, or another issue opened, until something that either changes the validation procedure or a way to opt out is merged? :)

Arksine, 2023-12-16:
This should now be possible with commit 3b53d95.  The check can be bypassed by setting the following option in `[file_manager]`:

```
check_klipper_config_path: False
```

eyJhb, 2023-12-16:
Perfect, thank you so much @Arksine ! Really appreciate the outcome of this :)

Tested and confirmed to work!