Arksine/moonraker #772: libgpiod v2 support

Kritnich, 2023-12-18:
### Is your feature request related to a problem? Please describe

libgpiod v2 was released [earlier this year](https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/tag/?h=v2.0). The new API is backwards incompatible. It would be good to support it sometime in the future since v1 will likely not be developed further.

### Describe the solution you'd like

Adjusting the gpio module behaviour so it reads the version from the installed libgpiod module and uses the API as such to add support for libgpiod v2 while maintaining compatibility with v1.

Arksine, 2023-12-24:
I took a look at this and am frankly perplexed as to why the `libgpiod` developers chose to release a new API with no backward compatibility.  As you indicate, this change requires applications that depend on `libgpiod` to support both APIs.

As such, I considered writing my own bindings for the gpio character device.  Fortunately for me, this has already been done in [python-periphery](https://github.com/vsergeev/python-periphery).  Commit  5d079d8c264e2927d02b2093e7a2e70846828769 migrates Moonraker's gpio component from `gpiod` to `python-periphery`.  Thanks.

Kritnich, 2023-12-28:
Thank you for your quick response and great effort. I was working on designing some bindings as well, but this is a much better solution. On my end the problem is solved with this commit.

closed 2023-12-28 by Kritnich