Arksine/moonraker #773: Creality Print support for metadata

jhemelaers-isense-development-be, 2023-12-21:
### Is your feature request related to a problem? Please describe

Metadata from Creality Print is not available. Parameters are a little bit different than the 'unknown' slicer

### Describe the solution you'd like

Generate a function for parsing the metadata in metadata.py

The content of Creality print is:

;FLAVOR:Marlin
;TIME:1720.57
;Filament used:4.98355m
;Layer Height:0.2
;MINX:36.127
;MINY:36.186
;MINZ:0.2
;MAXX:259.322
;MAXY:259.263
;MAXZ:9.6
;OuterWall Time:632.549
;InnerWall Time:184.264
;Skin Time:605.206
;Support Time:0
;SkirtBrim Time:31.7085
;Infill Time:99.1399
;Support Infills Time:0
;Combing Time:122.013
;Retraction Time:45.6892
;PrimeTower Time:0
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;Creality Print GCode Generated by CXEngine
;CRSLICE_GIT_HASH:6b8d75d
;Creality Print Version : V4.3.7.6619
;CXEngine Release Time:2023-09-20 15:19:09
;Gcode Generated Time:2023-12-21 12:19:03
;----------Machine Config--------------
;Machine Name:K1 Max
;Machine Height:300
;Machine Width:300
;Machine Depth:300
;Material Type:PLA
;Material Name:Generic-PLA_1.75
;Number of Extruders:1
;ExtruderParams[0] Nozzle Diameter:1000

; Pre-Processed for Cancel-Object support by preprocess_cancellation v0.2.0
; 1 known objects

Watch  carefully for the folllowing:

Filament used is in Creality Print without the space after the colon:
    def parse_filament_total(self) -> Optional[float]:
        return regex_find_float(
            r"; --- filament used: (%F) mm", self.footer_data
        )

Layer height is in Creality Print with capitale letter H(eight) and also no space after the colon

xz33xz33, 2024-01-09:
There is such an option https://github.com/ballaswag/moonraker/blob/61ae2cc8c8e4f41a7bc082591d4d06fabe4e8065/moonraker/components/file_manager/metadata.py

PJW58, 2024-02-17:
One should also be aware that Creality Print will also produce Klipper gcode, labeled as "Flavor: Creality OS".  For example the following generated for my Ender-3 V3 KE

;FLAVOR:Creality OS
;TIME:9185.19
;Filament used:8.57695m
;Material Diameter:1.75
;Material Density:1.24
;Filament Cost:0.511624
;Filament Weight:25.581177
;Filament Length:8.576949
;Layer height:0.1
;MINX:63.159
;MINY:68.563
;MINZ:0.2
;MAXX:156.841
;MAXY:151.437
;MAXZ:24
;OuterWall Time:2017.11
;InnerWall Time:1347.57
;Skin Time:1529.07
;Support Time:0
;SkirtBrim Time:159.102
;Infill Time:2394.21
;Support Infills Time:0
;Combing Time:1043.99
;Retraction Time:694.137
;PrimeTower Time:0
;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;;
;Creality Print GCode Generated by CXEngine
;CRSLICE_GIT_HASH:58ee74d
;Creality Print Version : V4.3.8.6984
;CXEngine Release Time:2023-12-04 18:16:29
;Gcode Generated Time:2024-02-17 12:47:59
;----------Machine Config--------------
;Machine Name:F005
;Machine Height:240
;Machine Width:220
;Machine Depth:220
;Material Type:PLA
;Material Name:Generic-PLA_1.75
;Number of Extruders:1
;ExtruderParams[0] Nozzle Diameter:1000