Arksine/moonraker #774: Authentication failure Shelly Plug Plus S

StinkendeHobbit, 2023-12-21:
### What happened

moonraker failes authenication with shelly plug plus s. 
https://www.shelly.com/en/products/shop/shelly-plus-plug-s

This results in the following log:
```
2023-12-17 21:14:05,815 [power.py:init_state()] - Device Init Error: shelly_plug
tornado.httpclient.HTTPClientError: HTTP 401: Unauthorized

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/user/moonraker/moonraker/components/power.py", line 451, in init_state
    state = await self._send_status_request()
  File "/home/user/moonraker/moonraker/components/power.py", line 1020, in _send_status_request
    res = await self._send_shelly_command("info")
  File "/home/user/moonraker/moonraker/components/power.py", line 1017, in _send_shelly_command
    return await self._send_http_command(url, command)
  File "/home/user/moonraker/moonraker/components/power.py", line 484, in _send_http_command
    response.raise_for_status(
  File "/home/user/moonraker/moonraker/components/http_client.py", line 363, in raise_for_status
    raise ServerError(msg, code) from self._error
moonraker.utils.ServerError: Error sending 'shelly' command: info
```

### Client

Mainsail

### Browser

Chrome

### How to reproduce

## config:
```c++
[power printer]
type: shelly
address: shelly_plug
password: example_password
off_when_shutdown: True
locked_while_printing: True
on_when_job_queued: True
restart_klipper_when_powered: True
restart_delay: 2
bound_service: klipper
```

### Additional information

I have had a the simmular issue with OctoPrint PSU control pluging. I managed to solve it with the following thread.
https://github.com/edekeijzer/OctoPrint-PSUControl-Shelly/issues/16

The issue seems to be related to the authentication method used by the plugin. As the new plug use a different method (HTTPDigestAuth). The provided solution has the following changes to handle this change in authentication method.

```python
if self.config['ng_device']:
    auth = HTTPDigestAuth(self.config['username'], self.config['password'])
else:
    auth = HTTPBasicAuth(self.config['username'], self.config['password'])
```

Arksine, 2023-12-22:
Thanks.  Indeed it appears that Gen 2 shelly devices have changed authentication methods.  For the time being I have noted in Moonraker's documentation that only Gen 1 devices support authentication and relabeled this as an feature request.   Hopefully an interested developer will add support for Gen 2 devices.

In the meantime, it should be possible to get the switch working over MQTT.  Alternatively, it is likely that Home Assistant supports Gen 2 shelly.

Elenedeath, 2024-08-11:
i found a workaround to use shelly plug S to turn printer on off
simply use the http type
```
#Shelly Plug S
[power printer]
type: http
on_url:http://192.168.1.45/relay/0?turn=on
off_url:http://192.168.1.45/relay/0?turn=off
```

Arksine, 2024-08-12:
Thanks.  This is the purpose of the generic http implementation...it should support nearly any switch.

That said, I suspect there is an error in your configuration.  The `request_template` is likely not necessary, as it seems that all parameters are provided in the query string.  The `response_template` is required, it needs process the response received from the device and evaluate to "on" or "off".

Hesi-Re, 2025-02-02:
I have the same issue. Elenedeath reported a good workaround, but it's probably incomplete, as Arksine suggested. What should we write in response_template field?