Arksine/moonraker #777: power function not working

mironfs, 2023-12-25:
### What happened

i updated klipper/moonraker/fluidd/os and now my button for powering on my printer from sidebar is missing

### Client

Fluidd

### Browser

Other or N/A

### How to reproduce

turn on printer

### Additional information

[moonraker (3).log](https://github.com/Arksine/moonraker/files/13767273/moonraker.3.log)
what i found: my button in fluidd went missing, later found fluidd hides it when there is not power option available
what i tried: i read log found this: ModuleNotFoundError: No module named 'periphery'
tried install/update periphery but didnt work for me
in the end i just replaced gpio.py with older version that doesnt call periphery in the start and now it works

i dont know if its somewhat related to issue #772 im not programmer

Arksine, 2023-12-25:
For some reason pip failed to install Moonraker's requirements during the update.  From your log:
```
2023-12-25 17:34:58,517 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/pip install -r /home/pi/moonraker/scripts/moonraker-requirements.txt) exited with return code 1
2023-12-25 17:35:00,944 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/pip install -r /home/pi/moonraker/scripts/moonraker-requirements.txt) exited with return code 1
2023-12-25 17:35:03,373 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/pip install -r /home/pi/moonraker/scripts/moonraker-requirements.txt) exited with return code 1
2023-12-25 17:35:03,373 [base_deploy.py:log_exc()] - Application moonraker: Error updating python requirements
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/update_manager/app_deploy.py", line 294, in _update_virtualenv
    return dep_info[distro_id]
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 565, in run_cmd
    def needs_service_restart(self, svc_name: str) -> bool:
utils.ServerError: Shell Command Error
```
The first 3 lines are the attempts to install the updated requirements file, which includes `python-periphery`.  Two things could have happened:
- Pip's servers were unreachable at the time, resulting in the errors
- The requirements file is corrupted

You can try to install it manually via SSH:
```
~/moonraker-env/bin/pip install -r ~/moonraker/scripts/moonraker-requirements.txt
```
If there is an error,  copy and paste it here.  If the requirements are successfully installed you can do a hard reset to restore `gpio.py` and restart moonraker:
```
cd ~/moonraker
git reset --hard origin/master
sudo service moonraker restart
```

ikonis, 2023-12-29:
I too lost GPIO control. the above steps did not fix it, nor throw any error

Arksine, 2023-12-29:
You will need to provide `moonraker.log` reproducing the issue.

ikonis, 2023-12-29:
Here it is.
[moonraker (2).log](https://github.com/Arksine/moonraker/files/13795835/moonraker.2.log)

ikonis, 2023-12-29:
`pi@klipper:~/moonraker/scripts $ sudo pip install -r ~/moonraker/scripts/moonraker-requirements.txt     Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring tornado: markers 'python_version == "3.7"' don't match your environment
Ignoring pillow: markers 'python_version == "3.7"' don't match your environment
Ignoring streaming-form-data: markers 'python_version == "3.7"' don't match your environment
Collecting tornado==6.3.3
  Downloading https://www.piwheels.org/simple/tornado/tornado-6.3.3-cp38-abi3-linux_armv7l.whl (426 kB)
     |████████████████████████████████| 426 kB 273 kB/s
Collecting pyserial==3.4
  Downloading https://www.piwheels.org/simple/pyserial/pyserial-3.4-py2.py3-none-any.whl (193 kB)
     |████████████████████████████████| 193 kB 136 kB/s
Collecting pyserial-asyncio==0.6
  Downloading https://www.piwheels.org/simple/pyserial-asyncio/pyserial_asyncio-0.6-py3-none-any.whl (7.5 kB)
Collecting pillow==10.0.1
  Downloading https://www.piwheels.org/simple/pillow/Pillow-10.0.1-cp39-cp39-linux_armv7l.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 113 kB/s
Collecting lmdb==1.4.1
  Downloading https://www.piwheels.org/simple/lmdb/lmdb-1.4.1-cp39-cp39-linux_armv7l.whl (249 kB)
     |████████████████████████████████| 249 kB 278 kB/s
Collecting streaming-form-data==1.13.0
  Downloading https://www.piwheels.org/simple/streaming-form-data/streaming_form_data-1.13.0-cp39-cp39-linux_armv7l.whl (337 kB)
     |████████████████████████████████| 337 kB 349 kB/s
Collecting distro==1.8.0
  Downloading https://www.piwheels.org/simple/distro/distro-1.8.0-py3-none-any.whl (20 kB)
Collecting inotify-simple==1.3.5
  Downloading https://www.piwheels.org/simple/inotify-simple/inotify_simple-1.3.5-py3-none-any.whl (9.4 kB)
Collecting libnacl==1.8.0
  Downloading https://www.piwheels.org/simple/libnacl/libnacl-1.8.0-py2.py3-none-any.whl (20 kB)
Collecting paho-mqtt==1.6.1
  Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.6.1-py3-none-any.whl (75 kB)
     |████████████████████████████████| 75 kB 197 kB/s
Collecting zeroconf==0.69.0
  Downloading https://www.piwheels.org/simple/zeroconf/zeroconf-0.69.0-cp39-cp39-manylinux_2_31_armv7l.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 45 kB/s
Collecting preprocess-cancellation==0.2.0
  Downloading https://www.piwheels.org/simple/preprocess-cancellation/preprocess_cancellation-0.2.0-py3-none-any.whl (18 kB)
Collecting jinja2==3.1.2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
     |████████████████████████████████| 133 kB 253 kB/s
Collecting dbus-next==0.2.3
  Downloading https://www.piwheels.org/simple/dbus-next/dbus_next-0.2.3-py3-none-any.whl (57 kB)
     |████████████████████████████████| 57 kB 150 kB/s
Collecting apprise==1.3.0
  Downloading https://www.piwheels.org/simple/apprise/apprise-1.3.0-py2.py3-none-any.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 100 kB/s
Collecting ldap3==2.9.1
  Downloading https://www.piwheels.org/simple/ldap3/ldap3-2.9.1-py3-none-any.whl (419 kB)
     |████████████████████████████████| 419 kB 273 kB/s
Collecting python-periphery==2.4.1
  Downloading https://www.piwheels.org/simple/python-periphery/python_periphery-2.4.1-py2.py3-none-any.whl (36 kB)
Requirement already satisfied: certifi in /usr/lib/python3/dist-packages (from apprise==1.3.0->-r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 19)) (2020.6.20)
Collecting requests-oauthlib
  Downloading https://www.piwheels.org/simple/requests-oauthlib/requests_oauthlib-1.3.1-py2.py3-none-any.whl (28 kB)
Collecting PyYAML
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-6.0.1-cp39-cp39-linux_armv7l.whl (45 kB)
     |████████████████████████████████| 45 kB 185 kB/s
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from apprise==1.3.0->-r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 19)) (2.25.1)
Collecting click>=5.0
  Downloading https://www.piwheels.org/simple/click/click-8.1.7-py3-none-any.whl (97 kB)
     |████████████████████████████████| 97 kB 156 kB/s
Collecting markdown
  Downloading https://www.piwheels.org/simple/markdown/Markdown-3.5.1-py3-none-any.whl (102 kB)
     |████████████████████████████████| 102 kB 132 kB/s
Collecting MarkupSafe>=2.0
  Downloading https://www.piwheels.org/simple/markupsafe/MarkupSafe-2.1.3-cp39-cp39-linux_armv7l.whl (23 kB)
Collecting pyasn1>=0.4.6
  Downloading https://www.piwheels.org/simple/pyasn1/pyasn1-0.5.1-py2.py3-none-any.whl (84 kB)
     |████████████████████████████████| 84 kB 167 kB/s
Collecting smart-open>=6.0
  Downloading https://www.piwheels.org/simple/smart-open/smart_open-6.4.0-py3-none-any.whl (57 kB)
     |████████████████████████████████| 57 kB 69 kB/s
[... 3 lines cut ...]
  Downloading https://www.piwheels.org/simple/async-timeout/async_timeout-4.0.3-py3-none-any.whl (5.7 kB)
Collecting importlib-metadata>=4.4
  Downloading importlib_metadata-7.0.1-py3-none-any.whl (23 kB)
Requirement already satisfied: zipp>=0.5 in /usr/lib/python3/dist-packages (from importlib-metadata>=4.4->markdown->apprise==1.3.0->-r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 19)) (1.0.0)
Collecting oauthlib>=3.0.0
  Downloading https://www.piwheels.org/simple/oauthlib/oauthlib-3.2.2-py3-none-any.whl (151 kB)
     |████████████████████████████████| 151 kB 102 kB/s
[... cut ...]
    Can't uninstall 'distro'. No files were found to uninstall.
Successfully installed MarkupSafe-2.1.3 PyYAML-6.0.1 apprise-1.3.0 async-timeout-4.0.3 click-8.1.7 dbus-next-0.2.3 distro-1.8.0 ifaddr-0.2.0 importlib-metadata-7.0.1 inotify-simple-1.3.5 jinja2-3.1.2 ldap3-2.9.1 libnacl-1.8.0 lmdb-1.4.1 markdown-3.5.1 oauthlib-3.2.2 paho-mqtt-1.6.1 pillow-10.0.1 preprocess-cancellation-0.2.0 pyasn1-0.5.1 pyserial-3.4 pyserial-asyncio-0.6 python-periphery-2.4.1 requests-oauthlib-1.3.1 smart-open-6.4.0 streaming-form-data-1.13.0 tornado-6.3.3 zeroconf-0.69.0`

ikonis, 2023-12-29:
pi@klipper:~/moonraker/scripts $ sudo pip install -r ~/moonraker/scripts/moonraker-requirements.txt     Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Ignoring tornado: markers 'python_version == "3.7"' don't match your environment
Ignoring pillow: markers 'python_version == "3.7"' don't match your environment
Ignoring streaming-form-data: markers 'python_version == "3.7"' don't match your environment
Collecting tornado==6.3.3
  Downloading https://www.piwheels.org/simple/tornado/tornado-6.3.3-cp38-abi3-linux_armv7l.whl (426 kB)
     |████████████████████████████████| 426 kB 273 kB/s
Collecting pyserial==3.4
  Downloading https://www.piwheels.org/simple/pyserial/pyserial-3.4-py2.py3-none-any.whl (193 kB)
     |████████████████████████████████| 193 kB 136 kB/s
Collecting pyserial-asyncio==0.6
  Downloading https://www.piwheels.org/simple/pyserial-asyncio/pyserial_asyncio-0.6-py3-none-any.whl (7.5 kB)
Collecting pillow==10.0.1
  Downloading https://www.piwheels.org/simple/pillow/Pillow-10.0.1-cp39-cp39-linux_armv7l.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 113 kB/s
Collecting lmdb==1.4.1
  Downloading https://www.piwheels.org/simple/lmdb/lmdb-1.4.1-cp39-cp39-linux_armv7l.whl (249 kB)
     |████████████████████████████████| 249 kB 278 kB/s
Collecting streaming-form-data==1.13.0
  Downloading https://www.piwheels.org/simple/streaming-form-data/streaming_form_data-1.13.0-cp39-cp39-linux_armv7l.whl (337 kB)
     |████████████████████████████████| 337 kB 349 kB/s
Collecting distro==1.8.0
  Downloading https://www.piwheels.org/simple/distro/distro-1.8.0-py3-none-any.whl (20 kB)
Collecting inotify-simple==1.3.5
  Downloading https://www.piwheels.org/simple/inotify-simple/inotify_simple-1.3.5-py3-none-any.whl (9.4 kB)
Collecting libnacl==1.8.0
  Downloading https://www.piwheels.org/simple/libnacl/libnacl-1.8.0-py2.py3-none-any.whl (20 kB)
Collecting paho-mqtt==1.6.1
  Downloading https://www.piwheels.org/simple/paho-mqtt/paho_mqtt-1.6.1-py3-none-any.whl (75 kB)
     |████████████████████████████████| 75 kB 197 kB/s
Collecting zeroconf==0.69.0
  Downloading https://www.piwheels.org/simple/zeroconf/zeroconf-0.69.0-cp39-cp39-manylinux_2_31_armv7l.whl (2.0 MB)
     |████████████████████████████████| 2.0 MB 45 kB/s
Collecting preprocess-cancellation==0.2.0
  Downloading https://www.piwheels.org/simple/preprocess-cancellation/preprocess_cancellation-0.2.0-py3-none-any.whl (18 kB)
Collecting jinja2==3.1.2
  Downloading https://www.piwheels.org/simple/jinja2/Jinja2-3.1.2-py3-none-any.whl (133 kB)
     |████████████████████████████████| 133 kB 253 kB/s
Collecting dbus-next==0.2.3
  Downloading https://www.piwheels.org/simple/dbus-next/dbus_next-0.2.3-py3-none-any.whl (57 kB)
     |████████████████████████████████| 57 kB 150 kB/s
Collecting apprise==1.3.0
  Downloading https://www.piwheels.org/simple/apprise/apprise-1.3.0-py2.py3-none-any.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 100 kB/s
Collecting ldap3==2.9.1
  Downloading https://www.piwheels.org/simple/ldap3/ldap3-2.9.1-py3-none-any.whl (419 kB)
     |████████████████████████████████| 419 kB 273 kB/s
Collecting python-periphery==2.4.1
  Downloading https://www.piwheels.org/simple/python-periphery/python_periphery-2.4.1-py2.py3-none-any.whl (36 kB)
Requirement already satisfied: certifi in /usr/lib/python3/dist-packages (from apprise==1.3.0->-r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 19)) (2020.6.20)
Collecting requests-oauthlib
  Downloading https://www.piwheels.org/simple/requests-oauthlib/requests_oauthlib-1.3.1-py2.py3-none-any.whl (28 kB)
Collecting PyYAML
  Downloading https://www.piwheels.org/simple/pyyaml/PyYAML-6.0.1-cp39-cp39-linux_armv7l.whl (45 kB)
     |████████████████████████████████| 45 kB 185 kB/s
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from apprise==1.3.0->-r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 19)) (2.25.1)
Collecting click>=5.0
  Downloading https://www.piwheels.org/simple/click/click-8.1.7-py3-none-any.whl (97 kB)
     |████████████████████████████████| 97 kB 156 kB/s
Collecting markdown
  Downloading https://www.piwheels.org/simple/markdown/Markdown-3.5.1-py3-none-any.whl (102 kB)
     |████████████████████████████████| 102 kB 132 kB/s
Collecting MarkupSafe>=2.0
  Downloading https://www.piwheels.org/simple/markupsafe/MarkupSafe-2.1.3-cp39-cp39-linux_armv7l.whl (23 kB)
Collecting pyasn1>=0.4.6
  Downloading https://www.piwheels.org/simple/pyasn1/pyasn1-0.5.1-py2.py3-none-any.whl (84 kB)
     |████████████████████████████████| 84 kB 167 kB/s
Collecting smart-open>=6.0
  Downloading https://www.piwheels.org/simple/smart-open/smart_open-6.4.0-py3-none-any.whl (57 kB)
     |████████████████████████████████| 57 kB 69 kB/s
[... 3 lines cut ...]
  Downloading https://www.piwheels.org/simple/async-timeout/async_timeout-4.0.3-py3-none-any.whl (5.7 kB)
Collecting importlib-metadata>=4.4
  Downloading importlib_metadata-7.0.1-py3-none-any.whl (23 kB)
Requirement already satisfied: zipp>=0.5 in /usr/lib/python3/dist-packages (from importlib-metadata>=4.4->markdown->apprise==1.3.0->-r /home/pi/moonraker/scripts/moonraker-requirements.txt (line 19)) (1.0.0)
Collecting oauthlib>=3.0.0
  Downloading https://www.piwheels.org/simple/oauthlib/oauthlib-3.2.2-py3-none-any.whl (151 kB)
     |████████████████████████████████| 151 kB 102 kB/s
[... cut ...]
    Can't uninstall 'distro'. No files were found to uninstall.
Successfully installed MarkupSafe-2.1.3 PyYAML-6.0.1 apprise-1.3.0 async-timeout-4.0.3 click-8.1.7 dbus-next-0.2.3 distro-1.8.0 ifaddr-0.2.0 importlib-metadata-7.0.1 inotify-simple-1.3.5 jinja2-3.1.2 ldap3-2.9.1 libnacl-1.8.0 lmdb-1.4.1 markdown-3.5.1 oauthlib-3.2.2 paho-mqtt-1.6.1 pillow-10.0.1 preprocess-cancellation-0.2.0 pyasn1-0.5.1 pyserial-3.4 pyserial-asyncio-0.6 python-periphery-2.4.1 requests-oauthlib-1.3.1 smart-open-6.4.0 streaming-form-data-1.13.0 tornado-6.3.3 zeroconf-0.69.0

Arksine, 2023-12-29:
`sudo pip install` is not the correct command.  Please run the command as shown above.

ikonis, 2023-12-29:
running it ANY way else produces nothing

ikonis, 2023-12-29:
pi@klipper:~ $ ~/moonraker-env/bin/pip install -r ~/moonraker/scripts/moonraker-requirements.txt
pi@klipper:~ $

Arksine, 2023-12-29:
Sounds like the virtualenv is corrupted.  Try the following:
```
cd ~
rm -rf moonraker-env
~/moonraker/scripts/install-moonraker.sh
```

ikonis, 2023-12-29:
That did it!

mironfs, 2024-01-01:
that worked for me too. weird thing is after restart it broke again. tried ~/moonraker/scripts/install-moonraker.sh second time and now it works even after restart. thanks. now im going to try update second printer

Arksine, 2024-01-01:
@mironfs Can you provide logs for the second printer?  Commit 54dc887 should have fixed dependency update issues due to the `zeroconf` build.   Its possible that something else caused it to fail depending on the system.

mironfs, 2024-01-01:
[moonraker (4).log](https://github.com/Arksine/moonraker/files/13805399/moonraker.4.log)

Arakon, 2024-01-08:
Just installed a fresh SD Card with MainsailOS, used kiauh to install Fluidd etc.
After updating Moonraker to latest, power button is grayed out, same error.
Ran [this](https://github.com/Arksine/moonraker/issues/777#issuecomment-1872328149) and it fixed it.

[moonraker.log](https://github.com/Arksine/moonraker/files/13866332/moonraker.log)

Arksine, 2024-01-08:
Thanks.  The version of Moonraker that ships with the last release of MainsailOS will need to be manually updated.  The `pip` updater in that version cannot correctly install Moonraker's requirements due to changes made to accommodate the now end-of-life Python 3.7.  The `pip` updater has since been fixed to handle this, and I plan to drop support for Python 3.7 later this year, likely when Python 3.8 is also EOL.

It shouldn't be necessary to remove the entire virtualenv in this case.  The following commands should perform a manual update:
```
cd ~/moonraker
git pull
./scripts/install-moonraker.sh
```

Arksine, 2024-02-04:
This should now be resolved.  Moonraker now checks its dependencies after an update, allowing it to install `python-periphery` on fresh Mainsail OS installations that have been updated.

closed 2024-02-04 by Arksine