Arksine/moonraker #778: Moonraker install/update high resource usage due to pip zeroconf install on low memory/embedded systems like Pi Zero 2 W

iointerrupt, 2023-12-26:
### Is your feature request related to a problem? Please describe

Moonraker install/update are extremely slow on systems like Pi Zero 2 W because pip Zeroconf installations require a compilation that causes high cpu and memory usage.  Any subsequent updates to library will result in a long hang during update.

### Describe the solution you'd like

Allow installation without zeroconf

### Additional information

During initial install of moonraker, I removed zeroconf from moonraker-requirements.txt.  However as I use fluidd, any updates to moonraker from the UI causes moonraker-requirements.txt to download and install zeroconf again.

Arksine, 2023-12-27:
Thanks.  I do have plans to address this.  For the time being I'll provide a brief explanation of what is going on and a workaround that should allow you to update without the long build time.

Last year the `zeroconf` dependency added cython support to gain a speed boost.  Unfortunately there are no prebuilt wheels for 64-bit versions of Raspberry Pi OS, so the cython source has to be built.  This process takes 20-30 minutes on a Pi 3 class CPU.

There is a still a pure python fallback implementation, however an environment variable needs to be set to skip the build.  You should be able to add the following to `~/printer_data/systemd/moonraker.env` to skip the cython build:

```
SKIP_CYTHON="y"
```

Then restart Moonraker.

iointerrupt, 2023-12-27:
Thank you @Arksine for the long winded and informative description and solution :-D.  That worked.

closed 2023-12-27 by iointerrupt

markwalkom, 2023-12-28:
Apologies for bumping a closed issue, but I did find that you can dramatically speed up the build of zerconf by increasing your swap size (I did 200MB on an rpi 3). This might help someone in the mean time!