Arksine/moonraker #779: Job stuck in "In Progress" state.

fjpanag, 2023-12-26:
### What happened

While printing I had a power failure that interrupted the job. The host computer (Raspberry Pi 4) was fully power cycled.

Upon restart, the interrupted job was stuck in the state "In Progress".

I printed successfully other jobs after the incident, which were correctly labelled as "In Progress" while printing and "Completed" when finished. But the problematic job is still in the wrong state.

What I would expect is moonraker to "recover" the interrupted job state to "Canceled" (or something similar) during the next boot.

Restarting moonraker or any other component, or fully rebooting the host didn't fix the issue.

### Client

Mainsail, Fluidd

### Browser

Firefox

### How to reproduce

These are steps to reproduce, I believe (although I didn't test them):

1. Start a print job.
2. While the print is in progress, restart the host machine (e.g. SSH and `reboot`).
3. Upon the next boot, the interrupted job should be stuck in "In Progress" state.

### Additional information

Here is the history, as presented by Mainsail:

[image]

Here is a copy of the log file:

[moonraker(1).log](https://github.com/Arksine/moonraker/files/13771882/moonraker.1.log)

Arksine, 2023-12-27:
Moonraker stores the job history in its database and modifies it twice: once when the job is started and again when its completed.   The power failure prevented the last update, hence the appearance of it being "stuck".  However this is purely cosmetic, it won't affect behavior in any way.

I don't know that Moonraker should be expected to attempt to correct historical data corrupted due to an abrupt shutdown, or at least I don't consider it a bug that it doesn't.  That said, I do think that in this case it can be done reliably with no repercussions, so its something that could be added as an enhancement.

Arksine, 2023-12-27:
This should now be resolved in commit 09b6a33ae4d85c0eb17c775e413480d5117987d7.

fjpanag, 2023-12-28:
@Arksine I totally understand that this is just cosmetic and not a functional issue.  
But since it "lies" about a job being in a state that it is not, it is great if it can be fixed.

If you are worried that this may cause other issues, maybe a "repair" script can be provided that the user will run manually when and if needed?

---

Nevertheless, I tried https://github.com/Arksine/moonraker/commit/09b6a33ae4d85c0eb17c775e413480d5117987d7 and I can confirm that it works as intended.
Thank you.

fjpanag, 2023-12-28:
While checking in Mainsail, I see that this `interrupted` state is not capitalized like the rest of them.
(See **C**anceled vs **i**nterrupted.)

[image]

I guess this is a front-end issue? That it doesn't recognize this new state maybe?
Shall I go ahead and create an issue on Mainsail, to make them aware of this change?

closed 2023-12-28 by fjpanag

Arksine, 2023-12-28:
The `status` reported by Moonraker is not capitalized, so I believe that you are correct that Mainsail needs to make a change.