Arksine/moonraker #780: You may have intended to navigate to one of Moonraker's front ends...

beebenutz, 2023-12-27:
### What happened

Installed Mainsail, Klipper, Moonraker, Crowsnest, and Klipperscreen via KIAUH. 
Mainsail connects perfectly via local:80 as intended. 
If I attempt to reach Moonraker via local:7125, I am greeted with a screen indication I am not connecting to the Moonraker "front-end". 
I have verified the port number and address. 

Please advise.

### Client

Mainsail

### Browser

Firefox

### How to reproduce

Install Mainsail, Klipper, Moonraker, Crowsnest, and Klipperscreen via KIAUH. 

Attempt to connect to USER.local:7125.

Receive the following greeting:

Welcome to Moonraker

You may have intended to navigate to one of Moonraker's front ends, if so check that you entered the correct port in the address bar.

### Additional information

[image]

Arksine, 2024-01-09:
This is Moonraker's landing page.  Its provides general information about the running instance and advises the user that if you intended to navigate to one of the front-ends you need to verify the port.  Typically they are served by nginx on port 80.

closed 2024-01-11 by beebenutz

ssna, 2024-01-13:
Same thing has just happend to me.  localhost takes me to first printer but no longer have menu to switch printers.  localhost:7125 or localhost:7126 tae me to the landing page.

blinkitogaming, 2024-03-22:
Same here as well. Few weeks, maybe one or two months ago I was able to reach any printer by specifying their ports rather 7125 or 7126 but since any of the past updates this takes me to that Moonraker welcome page.

I noticed this becouse I was setting up my Heimdall dashboard and it's configuration for showing special information about the app wasn't matching at all my printing status.

Arksine, 2024-03-22:
> Same here as well. Few weeks, maybe one or two months ago I was able to reach any printer by specifying their ports rather 7125 or 7126 but since any of the past updates this takes me to that Moonraker welcome page. But if I just go to the Mainsail's IP which listens on port 80 it goes to the printer that is instanced on port 7125...

Specifying the port in the browser navigation bar?  Moonraker's landing page was implemented years ago, its specifically intended to warn users that they have have may have typed in the URL and port, ie `http://myprinter.local:7125`, expecting to navigate to the frontend (Mainsail).

If you are typing in the port as part of Mainsail's configuration and it redirecting you to Moonraker's landing page then it may be a regression in Mainsail.

fbmoose48, 2024-05-03:
I have this same issue, what was the resolution?

fbmoose48, 2024-05-03:
> Nobody was able to provide an actual solution. My solution was to use a dedicated "raspberry pi zero w" on each of my machines. 

Appreciate the response, that's what I ended up doing as well - except with zero 2 w's to be able to keep webcams running.
Unfortunate, because these multi-port installs seem convenient. Although you may be right about relying on one instance of mainsail.

Arksine, 2024-05-04:
FWIW, it seems as if there may be a fundamental misunderstanding as to how Mainsail is intended to work with multiple instances.  Overall, the structure tends to be something like the following:

- nginx serves Mainsail on port 80.  In addition, nginx serves as a reverse proxy for the "primary" Moonraker instance, typically the one bound to port 7125.  Thus, all Moonraker HTTP API requests received over port 80 are forwarded to the instance on port 7125
- Additional instances of Moonraker are typically hosted on 7126, 7127, and so on.  This is configurable.
- If you navigate directly to any instance of Moonraker you will get the landing page shown in this issue.  That is, entering `http://my-printer.local:7125` in the nav bar of your browser will take you to the landing page.   This is the intended behavior and not a bug.   Moonraker has never served Mainsail or any other frontend, and the landing page has been implemented for several years.
- Navigating to the host on port 80 will take you to Mainsail, ie : `http://my-printer.local`.

Mainsail can be configured to connect to multiple instances of Moonraker.  How to do that is outside of the scope of Moonraker's issue tracker, however [Mainsail's Documentation](https://docs.mainsail.xyz/overview/settings/printers) is a good place to start.  If you are having trouble adding multiple instances, or it appears that it isn't working as intended, then I would suggest joining the Mainsail Discord linked in the docs for assistance.  You may also be able to create an issue on Mainsail's issue tracker.