Arksine/moonraker #781: Component 'history' failed to load with error

Guilouz, 2023-12-28:
### What happened

Since recent PR, when [history] is enabled in moonraker.conf , Moonraker report an issue.

[image]

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Add [history] in moonraker.conf

### Additional information

2023-12-28 01:43:42,917 [server.py:load_component()] - Component (authorization) loaded
2023-12-28 01:43:42,946 [server.py:load_component()] - Component (octoprint_compat) loaded
2023-12-28 01:43:43,009 [server.py:_initialize_component()] - Performing Component Post Init: [dbus_manager]
2023-12-28 01:43:43,045 [dbus_manager.py:component_init()] - Failed to get PolKit interface: The name org.freedesktop.PolicyKit1 was not provided by any .service files
2023-12-28 01:43:43,046 [server.py:_initialize_component()] - Performing Component Post Init: [file_manager]
2023-12-28 01:43:43,091 [server.py:_initialize_component()] - Performing Component Post Init: [machine]
2023-12-28 01:43:43,094 [machine.py:validation_init()] - Installation version in database up to date
2023-12-28 01:43:43,130 [machine.py:_find_public_ip()] - Detected Local IP: 192.168.0.37
2023-12-28 01:43:43,133 [server.py:_initialize_component()] - Performing Component Post Init: [proc_stats]
2023-12-28 01:43:43,134 [server.py:_initialize_component()] - Performing Component Post Init: [announcements]
2023-12-28 01:43:43,140 [server.py:_initialize_component()] - Performing Component Post Init: [webcam]
2023-12-28 01:43:43,141 [webcam.py:_set_default_host_ip()] - Default public webcam address set: http://192.168.0.37
2023-12-28 01:43:43,145 [webcam.py:_set_default_host_ip()] - Default public webcam address set: http://192.168.0.37
2023-12-28 01:43:43,146 [server.py:_initialize_component()] - Performing Component Post Init: [klippy_connection]
2023-12-28 01:43:43,148 [server.py:_initialize_component()] - Performing Component Post Init: [update_manager]
2023-12-28 01:43:43,151 [base_deploy.py:log_info()] - Git Repo moonraker: Stored pip version: 22.3.1
2023-12-28 01:43:43,161 [server.py:_initialize_component()] - Performing Component Post Init: [history]
2023-12-28 01:43:43,172 [server.py:_initialize_component()] - Component [history] failed post init
Traceback (most recent call last):
  File "/usr/data/moonraker/moonraker/moonraker/server.py", line 229, in _initialize_component
    await ret
  File "/usr/data/moonraker/moonraker/moonraker/components/history.py", line 93, in component_init
    jobs = await self.history_ns.get_batch(job_ids)
  File "/usr/data/moonraker/moonraker/moonraker/components/database.py", line 869, in get_batch
    return self.db.get_batch(self.namespace, keys)
  File "/usr/data/moonraker/moonraker/moonraker/components/database.py", line 447, in get_batch
    return self._run_command(self._get_batch_impl, namespace, keys)
  File "/usr/data/moonraker/moonraker/moonraker/components/database.py", line 204, in _run_command
    ret = func_wrapper()
  File "/usr/data/moonraker/moonraker/moonraker/components/database.py", line 199, in func_wrapper
    return command_func(*args)
  File "/usr/data/moonraker/moonraker/moonraker/components/database.py", line 458, in _get_batch_impl
    vals = cursor.getmulti(encoded_keys)
AttributeError: 'Cursor' object has no attribute 'getmulti'
2023-12-28 01:43:43,173 [server.py:add_warning()] - Component 'history' failed to load with error: 'Cursor' object has no attribute 'getmulti'
2023-12-28 01:43:43,173 [server.py:_initialize_component()] - Performing Component Post Init: [authorization]
2023-12-28 01:43:43,807 [git_deploy.py:log_repo_info()] - Git Repo moonraker Detected:

mfraser, 2023-12-28:
Cursor.getmulti was added to lmdb 1.1.0 and the version currently installed is 0.97.

2021-02-04 v1.1.0
* Add a Cursor.getmulti method with optional buffer support.

Arksine, 2023-12-28:
You need to attach the full log file.  Moonraker requires pylmdb version 1.4.1.

Arksine, 2023-12-28:
I have applied a workaround that should fix this error in commit f7d5f11cf8daa0c9891b979dcd293f7479701f74, however I am unable to test on the old version of lmdb.  If someone experiencing this issue can pull the latest version, confirm that it resolves the issue, and upload the resulting `moonraker.log` it would be appreciated.  I also added some logging for the current lmdb versions (both the library and the python bindings).

distrubi, 2023-12-28:
Updated and restarted moonraker

![image](https://github.com/Arksine/moonraker/assets
[moonraker.log](https://github.com/Arksine/moonraker/files/13787738/moonraker.log)
/47311848/77ae7bb8-1173-49d9-be69-9d958e5fda01)

Can confirm that the issue is gone.

Arksine, 2023-12-28:
Thank you for the log.  It looks like this issue affects K1 users.  During the update moonraker fails to install its requirements.  From the log:
```
2023-12-28 03:50:21,874 [shell_command.py:_check_proc_success()] - Command (/usr/data/moonraker/moonraker-env/bin/python -m pip install -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt) exited with return code -9
```
Something is killing `pip` before it can complete, somewhere between 2-4 minutes after its launched.   The long pip install time is likely related to #778, which should be resolved for future updates after commit 19422819da1d1ae9a8c3aa57fc790fa32844fa1d.  That said, the linked issue isn't what is killing the process prematurely.  The built in timeout is 20 minutes, so presumably it is an external process.

Arksine, 2023-12-28:
FWIW, I plan to move away from LMDB to sqlite3 soon.  Could someone with a K1 SSH into the machine, run the following command and post the output?
```
python -c "import sqlite3; print(sqlite3.sqlite_version)"
```
I'm curious to see if they have a build with sqlite preinstalled and what version it is.

distrubi, 2023-12-28:
root@K1Max-7651 /root [#] python3 -c "import sqlite3; print(sqlite3.sqlite_version)"
3.30.1

alenba90, 2023-12-28:
> FWIW, I plan to move away from LMDB to sqlite3 soon. Could someone with a K1 SSH into the machine, run the following command and post the output?
>
> ```
> python -c "import sqlite3; [...]

#root@K1Max-8605 /root [#] python -c "import sqlite3; print(sqlite3.sqlite_version)"
-sh: python: not found

h0licede, 2023-12-29:
hello,

any update for this? i still get this error. i have creality k1

Guilouz, 2023-12-29:
@h0licede It’s fixed with last commit.

h0licede, 2023-12-29:
> @h0licede It’s fixed with last commit.

im sorrry, what do i actaully do? i updated it through fluidd already. should i go to your installer script and reinstall it through there? sorry, im not sure

Guilouz, 2023-12-29:
You can just update moonraker from update manager in fluidd or mainsail

h0licede, 2023-12-29:
v0.8.0-239-gc4d15e40
thats my version of moonraker and i still have the error

distrubi, 2023-12-29:
Did you restart moonraker after updating?

h0licede, 2023-12-29:
i did firmware restart only. i will try to unplug the printer and restart if it helps

h0licede, 2023-12-29:
i can confirm that its fixed. i did unplugged the printer and started with no error

Hawk84r, 2023-12-30:
> FWIW, I plan to move away from LMDB to sqlite3 soon. Could someone with a K1 SSH into the machine, run the following command and post the output?
>
> ```
> python -c "import sqlite3; [...]

On latest firmware from Creality:
root@K1-BE4D /root [#] python3 -c "import sqlite3; print(sqlite3.sqlite_version)"
3.41.2

closed 2024-01-01 by Guilouz

Guilouz, 2024-01-29:
> Thank you for the log. It looks like this issue affects K1 users. During the update moonraker fails to install its requirements. From the log:
> 
> ```
> 2023-12-28 03:50:21,874 [shell_command.py:_check_proc_success()] - Command (/usr/data/moonraker/moonraker-env/bin/python -m pip install -r /usr/data/moonraker/moonraker/scripts/moonraker-requirements.txt) exited with return code -9
> ```
> 
> Something is killing `pip` before it can complete, somewhere between 2-4 minutes after its launched. The long pip install time is likely related to #778, which should be resolved for future updates after commit [1942281](https://github.com/Arksine/moonraker/commit/19422819da1d1ae9a8c3aa57fc790fa32844fa1d). That said, the linked issue isn't what is killing the process prematurely. The built in timeout is 20 minutes, so presumably it is an external process.

@Arksine The current version of moonraker requests to install a higher version of pillow than the K1 can handle.

For info, if moonraker-requirements.txt is changed to pillow==7.0.0 instead of 10.2.0 it's working.

Arksine, 2024-01-29:
That is unfortunate.  Pillow 10.0.2 fixes numerous critical security vulnerabilities and has some API changes.  In addition its necessary to support newer versions of Python.  I can't downgrade it to accommodate the K1.

It may be possible to modify the requirements file to exclude the Pillow install/update for the K1.  Pillow is only used in Moonraker's metadata parser, so it wouldn't affect general functionality.  Its possible that thumbnail creation will fail under some conditions with such an old version of Pillow.  I'm willing to look into it.

If someone can figure out how to build and install a more recent version of Linux on the K1 it will likely resolve some of the problems.  In October Python 3.8 will reach EOL status, and not long after that Moonraker will drop support for it too.