Arksine/moonraker #783: `python_wheels` subdir in `scripts/` breaks pdm wheel build

wlhlm, 2023-12-31:
### What happened

The latest commit (https://github.com/Arksine/moonraker/commit/54dc88725422c9f477a1dc75f20ef68e6c9f62b6) added a directory containing a wheel to `scripts/`. Was this intentional? Adding a subdir to `scripts/` breaks the `pdm_build_dist.py` script with the following error:

```
* Getting build dependencies for wheel...
* Building wheel...
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
    main()
  File "/usr/lib/python3.11/site-packages/pyproject_hooks/_in_process/_in_process.py", line 335, in main
    json_out['return_val'] = hook(**hook_input['kwargs'])
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
    return _build_backend().build_wheel(wheel_directory, config_settings,
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pdm/backend/__init__.py", line 54, in build_wheel
    return builder.build(
           ^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/pdm/backend/base.py", line 213, in build
    self.initialize(context)
  File "/usr/lib/python3.11/site-packages/pdm/backend/base.py", line 180, in initialize
    self.call_hook("pdm_build_initialize", context)
  File "/usr/lib/python3.11/site-packages/pdm/backend/base.py", line 138, in call_hook
    getattr(hook, hook_name)(context, *args, **kwargs)
  File "<snip>/moonraker/scripts/pdm_build_dist.py", line 77, in pdm_build_initialize
    shutil.copy2(str(item), str(scripts_dest))
  File "/usr/lib/python3.11/shutil.py", line 436, in copy2
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "/usr/lib/python3.11/shutil.py", line 256, in copyfile
    with open(src, 'rb') as fsrc:
         ^^^^^^^^^^^^^^^
IsADirectoryError: [Errno 21] Is a directory: '<snip>/moonraker/scripts/python_wheels'
```

### Client

Other

### Browser

Other or N/A

### How to reproduce

Build moonraker using `python -m build --wheel --no-isolation`,

Arksine, 2024-01-01:
Thanks.  This should be fixed in commit 84a8538597fe3f052f00808a0f174c71bc096566.

wlhlm, 2024-01-01:
Confirmed working. Thanks!

closed 2024-01-01 by wlhlm