Arksine/moonraker #785: `uvloop.install()` deprecated starting with Python 3.12

wlhlm, 2024-01-07:
### What happened

Just a heads up: I'm maintaining an unofficial moonraker package for Arch Linux and recently investigated the new optional `uvloop` (and `msgspec`) dependencies for potential performance improvements. During that I've noticed that moonraker's usage of `uvloop.install()` has been [deprecated](https://github.com/MagicStack/uvloop/blob/6c770dc3fbdd281d15c2ad46588c139696f9269c/uvloop/__init__.py#L28-L37) starting with (C)Python 3.12. This does not seem urgent (Arch itself hasn't even switched to py3.12 yet), just an FYI.

### Client

Other

### Browser

Other or N/A

### How to reproduce

*N/A*

Arksine, 2024-01-08:
Thanks, this should be resolved in commit 27a02952184e8d253510073704d0e10f2807e06b.

wlhlm, 2024-01-08:
Great!

closed 2024-01-08 by wlhlm