Arksine/moonraker #787: Inverted GPIO power pin is turned on briefly every time Moonraker loads

Arakon, 2024-01-09:
### What happened

I have a Lerdge Power Monitor relay which requires an active low signal to turn on. It's connected to GPIO23 on the raspberry pi.
The config looks like this:
```
[power printer]
type: gpio
pin: !gpiochip0/gpio23
off_when_shutdown: True
restart_klipper_when_powered: true
initial_state: off

```
Whenever Moonraker starts up, after powering on the Pi or restarting the Moonraker service, it will briefly turn on the relay and then back off. However, if the pin is NOT inverted, the relay turns on as soon as Moonraker starts up, and stays on (which is expected), and does not briefly turn it off. 
Note that the relay doesn't toggle during Pi startup or when Moonraker is not launched, so this is definitely triggered by Moonraker.
I also tried with GPIO4 instead, with the exact same behavior.
On another printer, with an active high relay, and non-inverted GPIO pin, the relay is only triggered when commanded.

Expected behavior:
Moonraker keeps the default GPIO state until actively instructed otherwise.
[moonraker.log](https://github.com/Arksine/moonraker/files/13878332/moonraker.log)

### Client

Fluidd

### Browser

Firefox

### How to reproduce

- Attach an active low triggered relay to a GPIO
- Set up Moonraker to switch it as a power toggle with an inverted GPIO
- Start or Restart Moonraker and notice that the GPIO is triggered briefly on launch

Arksine, 2024-01-10:
Thanks.  This should be resolved in commit ea6df41f05a11622f4377bf26a09a08edb773378.

Arakon, 2024-01-10:
Fixed. Thanks for the quick response!

closed 2024-01-11 by Arksine

Arakon, 2024-01-17:
One small addition: When shutting down the host, the pin does not turn off. Is this something that can be done on the moonraker side?

Arksine, 2024-01-18:
Generally Moonraker does not change power device state when it shuts down for any of its implementations.  It does release the gpio line and its possible that `libgpiod` reset the pin's state to whereas `python-periphery` does not.  I'm willing to look into it, but I'm also hesitant to make a change here.  Consider that if your pin is inverted then resetting the state may power on a relay.

Arksine, 2024-01-18:
I did look into this and this behavior hasn't changed with `python-periphery` as libgpiod also leaves the line state unchanged after release.  I don't think it would be appropriate for Moonraker to make assumptions about what the line state should be when the service is stopped, so I'm inclined to leave the behavior as is.