Arksine/moonraker #788: Cannot shutdown or reboot host

xiaojie0815, 2024-01-11:
### What happened

What happened?
When trying to shutdown or reboot the host system for the GUI, it shows the following error in the log and just ignores the command:
`2024-01-11 21:24:10,561 [common.py:build_error()] - JSON-RPC Request Error - Requested Method: machine.reboot, Code: -31000, Message: Interactive authentication required.`

KlipperScreen.log  As shown below
[image]

I tried the following commands but none of them work
the first method:
cd ~/moonraker/scripts
./set-policykit-rules.sh
sudo service moonraker restart
The second method:
cd ~/moonraker/scripts
./install-moonraker.sh

### Client

Fluidd, KlipperScreen

### Browser

Chrome

### How to reproduce

Go to system -> shutdown -> host system

No error, and host shutdown.

xiaojie0815, 2024-01-11:
system message
[image]

Arksine, 2024-01-11:
It looks like you are trying to run Moonraker on a smartphone.  While this may work for general purpose usage, the distribution could be missing packages Moonraker depends on such as `dbus` and `policykit` for its extended functionality.  Alternatively it may not use `systemd` as the init manager, in which case Moonraker will be unable to perform actions such as `shutdown` and `reboot`.  If its running systemd but missing the aforementioned packages you could try to [configure the provider option](https://moonraker.readthedocs.io/en/latest/configuration/#machine). in `[machine]` to `systemd_cli`.

You could attach `moonraker.log` for me to get a more detailed look at the system and I'll see if I can provide further guidance, but I suspect the answer is that Moonraker will be unable to perform a shutdown in your environment.

xiaojie0815, 2024-01-12:
> 您似乎正在尝试在智能手机上运行 Moonraker。虽然这可能适用于一般用途，但发行版可能会缺少 Moonraker 所依赖的软件包`dbus`以及`policykit`其扩展功能。或者，它可能不用作`systemd`init 管理器，在这种情况下，Moonraker 将无法执行诸如`shutdown`和之类的操作`reboot`。如果它正在运行 systemd 但缺少上述软件包，您可以尝试在to中[配置提供程序选项](https://moonraker.readthedocs.io/en/latest/configuration/#machine)。`[machine]``systemd_cli`
> 
> 您可以附上附件`moonraker.log`让我更详细地了解系统，我会看看是否可以提供进一步的指导，但我怀疑答案是 Moonraker 将无法在您的环境中执行关闭。

[moonraker.log](https://github.com/Arksine/moonraker/files/13919024/moonraker.log)
This is the log file I generated. Please take a look. Thank you very much for your help.

xiaojie0815, 2024-01-13:
@Arksine 

I compared the software packages that can be installed on the Debian system.

I uninstalled the following packages
1.dbus-bin
2.dbus-daemon
3. dbus-session-bus-common
4. dbus-system-bus-common

Then install these two packages
1. sudo apt install x11-utils
2. sudo apt install xdg-utils

After uninstalling and reinstalling Moonraker, the shutdown and restart commands now work normally. I don't know the specific reason yet.

Arksine, 2024-05-08:
As this discussion seems to have concluded I'm going to close this.  It appears the cause is distro related.

closed 2024-05-08 by Arksine