Arksine/moonraker #794: Installed Klipper and Moonraker and fluuid thru kiauh - moonraker reports it cant start - 502 bad gateway

soundtechscott, 2024-01-20:
### What happened

I have removed elegoo stuff from an image and installed standard klipper instead of the elegoo messed up garbage version of klipper
process involves running a script to remove all elegoo customizations and basically bring the image back to a standard ish version of armbian
then you install klipper, moonraker, and fluidd thru kiauh
update your config file for the printer and it usually just works

but this version doesnt seem to work for me
this is on an Elegoo Neptune 4, with a makerbase skipr based board running Armbian 22.05.0-trunk  with bleeding edge Linux 6.1.11-rockchip64 
 
the username is "mks" (not pi)

I get this error

 moonraker.service - API Server for Klipper SV1
   Loaded: loaded (/etc/systemd/system/moonraker.service; enabled; vendor preset
   Active: activating (auto-restart) (Result: exit-code) since Fri 2024-01-19 22
     Docs: https://moonraker.readthedocs.io/
  Process: 3636 ExecStart=/home/mks/moonraker-env/bin/python $MOONRAKER_ARGS (co
 Main PID: 3636 (code=exited, status=1/FAILURE)
 
a little further digging into logs shows that it has file permission issues....
I had to chmod diretcories and files for moonraker stuff and give them read/write permissions for it to be happy

in my case
/home/mks/printer_data/database/ and files

### Client

Fluidd

### Browser

Chrome

### How to reproduce

I guess you could reproduce it by buying a neptune 4 with the v2.o board on it and trying to use the OpenNept4une script to remove elegoo's old klipper version and customizations and then use the de_elegoo_cleanser.sh file

### Additional information

I dont have a failed moonraker log, but every error it gave me said it didnt have permissions to various files or the database files

soundtechscott, 2024-01-20:
heres the latest log of the error

-------------------- Log Start | Sat Jan 20 04:56:55 2024 --------------------
data_path: /home/mks/printer_data
is_default_data_path: False
config_file: /home/mks/printer_data/config/moonraker.conf
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: f2e4bc29fd8a457ea30a6570f75bd797
unix_socket_path: /home/mks/printer_data/comms/moonraker.sock
software_version: v0.8.0-267-g66de18f
git_branch: master
git_remote: origin
git_repo_url: https://github.com/Arksine/moonraker.git
modified_files: []
unofficial_components: []
log_file: /home/mks/printer_data/logs/moonraker.log
python_version: 3.7.3 (default, Oct 11 2023, 09:51:27)  [GCC 8.3.0]
launch_args: /home/mks/moonraker-env/bin/python /home/mks/moonraker/moonraker/moonraker.py -d /home/mks/printer_data
msgspec_enabled: False
uvloop_enabled: False
2024-01-19 22:56:55,253 [confighelper.py:read_file()] - Configuration File '/home/mks/printer_data/config/moonraker.conf' parsed, total size: 466 B
2024-01-19 22:56:55,254 [server.py:add_log_rollover_item()] - 
#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /home/mks/printer_data/comms/klippy.sock

[authorization]
trusted_clients = 
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	192.168.1.0/24
	FE80::/10
	::1/128
cors_domains = 
	*.lan
	*.local
	*://localhost
	*://localhost:*
	*://my.mainsail.xyz
	*://app.fluidd.xyz

[octoprint_compat]

[history]

[update_manager]
channel = dev
refresh_interval = 168

#################################################################
All Configuration Files:
/home/mks/printer_data/config/moonraker.conf
#################################################################
2024-01-19 22:56:55,453 [server.py:load_component()] - Component (secrets) loaded
2024-01-19 22:56:55,466 [server.py:load_component()] - Component (template) loaded
2024-01-19 22:56:55,482 [server.py:load_component()] - Component (klippy_connection) loaded
2024-01-19 22:56:55,770 [server.py:load_component()] - Component (application) loaded
2024-01-19 22:56:55,810 [server.py:load_component()] - Component (websockets) loaded
2024-01-19 22:56:55,903 [server.py:load_component()] - Component (dbus_manager) loaded
2024-01-19 22:56:55,914 [server.py:load_component()] - Unable to load component: (database)
Traceback (most recent call last):
  File "/home/mks/moonraker/moonraker/server.py", line 291, in load_component
    component = load_func(config)
  File "/home/mks/moonraker/moonraker/components/database.py", line 956, in load_component
    return MoonrakerDatabase(config)
  File "/home/mks/moonraker/moonraker/components/database.py", line 102, in __init__
    self.database_path, map_size=MAX_DB_SIZE, max_dbs=MAX_NAMESPACES
lmdb.ReadonlyError: /home/mks/printer_data/database: Permission denied
2024-01-19 22:56:55,915 [server.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/home/mks/moonraker/moonraker/server.py", line 647, in main
    server.load_components()
  File "/home/mks/moonraker/moonraker/server.py", line 255, in load_components
    self.load_component(config, component)
  File "/home/mks/moonraker/moonraker/server.py", line 291, in load_component
    component = load_func(config)
  File "/home/mks/moonraker/moonraker/components/database.py", line 956, in load_component
    return MoonrakerDatabase(config)
  File "/home/mks/moonraker/moonraker/components/database.py", line 102, in __init__
    self.database_path, map_size=MAX_DB_SIZE, max_dbs=MAX_NAMESPACES
lmdb.ReadonlyError: /home/mks/printer_data/database: Permission denied
2024-01-19 22:56:55,916 [server.py:main()] - Server Shutdown

soundtechscott, 2024-01-20:
[moonraker.log](https://github.com/Arksine/moonraker/files/13996878/moonraker.log)

Arksine, 2024-01-20:
Unfortunately there isn't much I can do for this . While generally KIAUH works well and I recommend it, its 3rd party software so I can't explain why it would create the `printer_data` folder without the appropriate permissions on a fresh install.  If you attempted to copy the folder or the files within it from another installation then that would perhaps explain the permission issues.  I suspect the only solution will be to modify the permissions manually as you mentioned in the OP.

Arksine, 2024-02-04:
As no one has been able to provide further feedback I'm going to close this.  If you are still having issues I recommend asking for assistance on the Klipper Discourse.

closed 2024-02-04 by Arksine