Arksine/moonraker #795: Loading gcode files causes fluidd to lose connection to moonraker

samwiseg0, 2024-01-21:
### What happened

On the latest version it appears there is some timeout that is triggered that causes fluidd to lose communication to moonraker. 

This happens when trying to load the list of gcode files. When opening a folder where the gocde files are the preview image may load on a few gcode files and then the interface hangs and becomes unresponsive. 

To test I rolled back to commit, https://github.com/Arksine/moonraker/commit/6d8cb762ffca1bd151d98e0cec5b66b966965014 and everything works as expected. 

### Client

Fluidd

### Browser

Chrome, Safari, Other or N/A

### How to reproduce

Run the latest version. I am using authz but I am unsure if that is related to this issue. Navigate to jobs and open a folder containing gcode files and the interface will become unresponsive. 

### Additional information

[moonraker-5.log](https://github.com/Arksine/moonraker/files/14002257/moonraker-5.log)

Seeing this repeatedly in the logs. 
```
2024-01-21 12:08:31,111 [proc_stats.py:_watchdog_callback()] - EVENT LOOP BLOCKED: 20.31 seconds, total blocked count: 21
```

Arksine, 2024-01-22:
Thanks.  I'm working on tracking this down but thus far have been unable to reproduce.  Can you add the following line to `~/printer_data/systemd/moonraker.env`?

```
MOONRAKER_VERBOSE_LOGGING="y"
```

Then restart Moonraker, repro the issue, and upload the log.  This will create large logs, but will hopefully help me track down where the blocking issue is occurring (and how its related to authorization, it really shouldn't be).

Arksine, 2024-01-22:
Update:

Even though I can't reproduce I think I have found the source of the problem.  There is a potential blocking call to `socket.getfqdn` in the `authorization` module that was generally bypassed previously.  Its a call that is very rarely reached and normally it returns immediately.  I'll work on a fix.  You can hold off on the verbose logging right now, as I am reasonably confident that this is the problem.

samwiseg0, 2024-01-22:
> Update:
>
>
>
> Even though I can't reproduce I think I have found the source of the problem.  There is a potential blocking call to `socket.getfqdn` in the `authorization` module that was generally [...]

This makes sense. I discovered late last night that my first DNS server was unresponsive so all queries were timing out on the first and then heading to the second DNS server.

Arksine, 2024-01-22:
This should be resolved in d1f97f2658a4e003da95c131c87ce4c59e86de8f.  In addition to running `getfqdn()` in a thread to prevent blocking, it will now only be called if a domian name is configured in `trusted_clients`.

samwiseg0, 2024-01-22:
Awesome! I will simulate the DNS failure and test before and after this commit and report back. 

Thanks again for looking into this so quickly!

samwiseg0, 2024-01-23:
As best as I could test. I am mostly confident this resolved the issue. 

When doing my testing I was not able to have fluidd lose connection entirely but it would freeze for a few seconds loading gcode file images. After your change there was no delay. With or without the DNS server being down.