Arksine/moonraker #800: Regularly get stuck at "Initializing..."

esb7, 2024-01-26:
### What happened

I regularly get stuck at "Initializing..." when accessing mainsail from a web browser on my Mac.  Sometimes it works and other times it fails and will continue to fail even through reboots of the rPi hosting mainsail.  It would occur on different web browsers as well.

### Client

Mainsail

### Browser

Chrome, Safari

### How to reproduce

Use `mainsailos.local` to access the printer web UI.

### Additional information

The issue seems caused by "unauthorized" errors that show up in the moonraker log:

> `2024-01-20 18:50:33,939 [websockets.py:on_close()] - Websocket Closed: ID: 2822839248 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 8.53
2024-01-20 18:50:34,137 [websockets.py:prepare()] - Websocket Failed Authentication: HTTP 401: Unauthorized (Unauthorized)
2024-01-20 18:50:34,140 [app.py:log_request()] - 101 GET /websocket (fd70:616c:c5db:914e:18f6:e20b:5620:9152) [No User] 23.93ms
`

I noticed that the failed authentications all had a reported ipv6 address and on further investigation were all in a range outside of the allowed range in the `trusted clients:` section in `moonraker.conf`.  The primary / default trusted ipv6 range is `FE80::/10`.  However that appear to exclude other local-only address ranges (including the FD70 my device had).  I believe the default should be: `FC00::/7`.

For anybody that hits this, you can get past it by adding the following to the `trusted clients:` section of the `moonraker.conf` file:
`    FC00::/7`

Arksine, 2024-01-26:
Thanks.  For the record, Moonraker does not provide default values for the `trusted_clients` option, its entirely up to users to configure.  I wouldn't want to inadvertently authorize IPs the user is not aware of out of the box, including local blocks.

I suspect you are referring to the configuration provided by MainsailOS.  If you believe that FC00::/7 should be included in the list of trusted clients you may wish to create an issue on their repo.  Alternatively you can probably talk to someone in their discord about it.

Arksine, 2024-05-08:
A this discussion has concluded I'm going to close this.  Thanks.

closed 2024-05-08 by Arksine