Arksine/moonraker #806: Moonraker v0.8.0-294-g61ea8603 network issue after update.

DuckersMcQuack, 2024-02-03:
### What happened

After i updated klipper and moonraker, i now get 20-50% chance of failed upload. Worked perfectly fine before the update.

### Client

Mainsail

### Browser

Other or N/A

### How to reproduce

Run cura 4.12, connect to newest updated moonraker and klipper , upload a gcode through cura, and it will have either luck uplading fine the first go, or it can go 2-6 retries until it succeds. It was success every upload before the latest update.

[image]

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/14152336/moonraker.log)
[klippy (1).log](https://github.com/Arksine/moonraker/files/14152337/klippy.1.log)

Arksine, 2024-02-04:
This looks like a duplicate of #736.  Make sure you are using the latest version of the moonraker plugin.

Arksine, 2024-05-08:
As this discussion seems to have concluded I'm going to close this.

closed 2024-05-08 by Arksine