Arksine/moonraker #807: Moonraker Corrupt error on initial component update

ankonaskiff17, 2024-02-03:
### What happened

I have installed all components via the Raspberry Pi Imager process, launch Mainsail via .local. When Mainsail launches in browser tab the Update Manager is empty but tells me it has not been initialized and to hit refresh so I hit what I assume is refresh in Update Manager and it populates with the expected files, crowsnest, klipper, mainsail, mainsail-config, moonraker, sonar and system.
Have done this initial setup via Pi Imager twice and gotten multiple items indicating Corrupt, did Hard Recovery in both cases and in first case everything cleaned up and indicated okay.
The most recent time klipper and moonraker showed Corrupt with klipper resolving after Hard Recovery but moonraker has failed after two attempts at Hard Recovery.  Things are moving very quickly but it appears that process is having trouble accessing git. 
Am going to pull, reformat and make another attempt at getting a clean installation on my micro-SD card.
Attached in the moonraker log which I rolled over prior to refreshing Update Manager

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Launch Raspberry Pi Imager (V 1.8.4)
Choose Pi 4
Choose purpose built OS
Choose 3D printer
Choose MainsailOS 64 bit*  The 32 bit MainsalOS option has been removed from Pi Imager 

### Additional information

O
Owner: mainsail-crew
Repository Name: sonar
Path: /home/pi/sonar
Remote: origin
Branch: main
Remote URL: https://github.com/mainsail-crew/sonar.git
Recovery URL: https://github.com/mainsail-crew/sonar.git
Current Commit SHA: eacc498a52f79292063b6f9b0e0c191c34c5dc09
Upstream Commit SHA: eacc498a52f79292063b6f9b0e0c191c34c5dc09
Current Version: v0.1.3-1-geacc498a
Upstream Version: v0.1.3-1-geacc498a
Rollback Commit: eacc498a52f79292063b6f9b0e0c191c34c5dc09
Rollback Branch: main
Rollback Version: v0.1.3-1-geacc498a
Is Dirty: False
Is Detached: False
Is Shallow: False
Commits Behind Count: 0
Diverged: False
2024-02-03 16:08:54,848 [base_deploy.py:log_info()] - PackageKit: Next refresh in: 6 Days, 23 Hours, 57 Minutes, 6 Seconds
2024-02-03 16:08:54,848 [base_deploy.py:log_info()] - Git Repo moonraker: Next refresh in: 6 Days, 23 Hours, 59 Minutes, 55 Seconds
2024-02-03 16:08:54,848 [base_deploy.py:log_info()] - Git Repo klipper: Next refresh in: 6 Days, 23 Hours, 58 Minutes, 42 Seconds
2024-02-03 16:08:54,848 [base_deploy.py:log_info()] - Web Client mainsail: Next refresh in: 6 Days, 23 Hours, 57 Minutes, 10 Seconds
2024-02-03 16:08:54,848 [base_deploy.py:log_info()] - Git Repo mainsail-config: Next refresh in: 6 Days, 23 Hours, 57 Minutes, 11 Seconds
2024-02-03 16:08:54,849 [base_deploy.py:log_info()] - Git Repo crowsnest: Next refresh in: 6 Days, 23 Hours, 57 Minutes, 13 Seconds
2024-02-03 16:08:54,849 [base_deploy.py:log_info()] - Git Repo sonar: Next refresh in: 6 Days, 23 Hours, 57 Minutes, 14 Seconds
2024-02-03 16:08:54,850 [confighelper.py:create_backup()] - Backing up last working configuration to '/home/pi/printer_data/config/.moonraker.conf.bkp'
2024-02-03 16:08:54,854 [extensions.py:start_unix_server()] - Creating Unix Domain Socket at '/home/pi/printer_data/comms/moonraker.sock'
2024-02-03 16:08:54,856 [server.py:start_server()] - Starting Moonraker on (0.0.0.0, 7125), Hostname: ender3pro
2024-02-03 16:08:54,858 [application.py:listen()] - SSL Certificate/Key not configured, aborting HTTPS Server startup
2024-02-03 16:08:55,111 [klippy_connection.py:_do_connect()] - Klippy Connection Established
2024-02-03 16:08:55,273 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.1.13
2024-02-03 16:08:55,275 [application.py:log_request()] - 101 GET /websocket (192.168.1.13) [_TRUSTED_USER_] 3.13ms
2024-02-03 16:08:55,275 [websockets.py:open()] - Websocket Opened: ID: 548621203056, Proxied: True, User Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/121.0.0.0 Safari/537.36, Host Name: ender3pro.local
2024-02-03 16:08:55,283 [websockets.py:_handle_identify()] - Websocket 548621203056 Client Identified - Name: mainsail, Version: 2.9.1, Type: web
2024-02-03 16:08:55,423 [server.py:add_log_rollover_item()] - Klipper Version: v0.12.0-102-g9f41f53c
2024-02-03 16:08:55,440 [klippy_connection.py:_request_initial_subscriptions()] - Webhooks Subscribed
2024-02-03 16:08:55,441 [klippy_connection.py:_request_initial_subscriptions()] - GCode Output Subscribed
2024-02-03 16:08:55,443 [klippy_connection.py:_check_ready()] - 
Unable to open config file /home/pi/printer_data/config/printer.cfg

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
[... 101 lines cut ...]
    return await self.run_in_executor(
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution
2024-02-03 16:39:13,408 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<function _HTTPConnection.__init__.<locals>.<lambda> at 0x7fbc544790>, <Task finished name='Task-13556' coro=<_HTTPConnection.run() done, defined at /home/pi/moonraker-env/lib/python3.9/site-packages/tornado/simple_httpclient.py:290> exception=gaierror(-3, 'Temporary failure in name resolution')>)
Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/ioloop.py", line 750, in _run_callback
    ret = callback()
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/simple_httpclient.py", line 287, in <lambda>
    gen.convert_yielded(self.run()), lambda f: f.result()
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/simple_httpclient.py", line 340, in run
    stream = await self.tcp_client.connect(
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/tcpclient.py", line 269, in connect
    addrinfo = await self.resolver.resolve(host, port, af)
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/netutil.py", line 433, in resolve
    for fam, _, _, _, address in await asyncio.get_running_loop().getaddrinfo(
  File "/usr/lib/python3.9/asyncio/base_events.py", line 856, in getaddrinfo
    return await self.run_in_executor(
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/lib/python3.9/socket.py", line 953, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution

Arksine, 2024-02-04:
You are having network/DNS resolution issues.  When Moonraker attempts to fetch your repos its failing.  Aside from that its important not to interrupt Moonraker when its trying to perform a git operation.  This can lead to "corruption".

Fix your network issues and things should correct themselves after refresh.

el-quique, 2024-02-04:
[ankonaskiff17](https://github.com/ankonaskiff17) Hello. Try uninstalling moonraker (previously backup config) and then reinstall

ankonaskiff17, 2024-02-08:
How are you determining that I am having network problems? I probably didn’t get in enough detail but these corrupt file indications are not when popping up when I am updating the various components.
This is what I am confronted with upon the initial install of Mainsail. So I type in ender3pro.local in a new Chrome tab and it launches with those component’s indicating corrupt. When I would click on Hard Recovery for each component that indicated Corrupt and that action would clear the Corrupt alert. 
 
I am not going from a green text block telling me to update and when I select Update getting a corrupt error 
Corrupt is first thing I would see upon first browser launch.
I pulled the SD card, formatted it and went through the Pi Imager process a second time and it came up with no Corrupt indications so I ran with it.
 

 
 
Wayne Wells
email: ***@***.***
Cell: (803) 553-6091
 
10682 Two Notch Rd.
Apt. 7107
Elgin, SC 29045

Arksine, 2024-02-08:
This is from your op:

> The most recent time klipper and moonraker showed Corrupt with klipper resolving after Hard Recovery but moonraker has failed after two attempts at Hard Recovery. Things are moving very quickly but it appears that process is having trouble accessing git.

The reason it cannot successfully perform the recovery is a DNS failure.  From the log snippet there are several DNS resolution errors:

```
socket.gaierror: [Errno -3] Temporary failure in name resolution
```

Most likely this is why the repos are reporting as corrupt, and it is definitely the reason hard recovery failed.

leandropaolo1, 2024-04-28:
Do you happen to have your VPN on? Try turning it off and then hard reset, then download again.

Arksine, 2024-05-08:
As this discussion has concluded I'm going to close this.

closed 2024-05-08 by Arksine