Arksine/moonraker #808: Moonraker host shutdown not working

Gizzzle, 2024-02-07:
### What happened

When trying to shutdown the Raspberry CM4 module on my Manta M8P board, nothing happens. In the moonraker.log the following error comes up:

[common.py:build_error()] - JSON-RPC Request Error - Requested Method: machine.shutdown, Code: 500, Message: Error running shell command: 'sudo systemctl poweroff'

### Client

Mainsail

### Browser

Firefox

### How to reproduce

In mainsail press the host shutdown button

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/14201339/moonraker.log)

Arksine, 2024-02-07:
You have the `systemd_cli` provider configured in `[machine]`.  When you do this the user that owns the Moonraker process needs "no password" sudo permission for Moonraker to restart.  FWIW, you also have packagekit disabled in the `update_manager`.  APT needs the "no password" permission to install packages, so this too will fail if you attempt to do so.

I would recommend using the default `systemd_dbus` provider.  The polkit permissions should be installed already, but if not the script in `~/moonraker/scripts/set-policykit-rules.sh` can add them.  If you don't want to do this, you need to give the user `no password` access.

Arksine, 2024-05-08:
As this discussion has concluded I'm going to close this.

closed 2024-05-08 by Arksine