Arksine/moonraker #810: Spoolman startup error

Horgul123, 2024-02-13:
### What happened

I know that there is another issue regarding spoolman, but as I understood these issue should be solved, so I decided to open a new one

After starting my Creality K1 it fails to connect to spoolman. When I restart moonraker the connection works. So it looks like a moonraker issue, as spoolman is installed on another server.

[image]

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Start moonraker/Mailsail

### Additional information

Moonraker Version: v0.8.0-310-g67c98f6a
[moonraker (1).log](https://github.com/Arksine/moonraker/files/14257267/moonraker.1.log)

Arksine, 2024-02-13:
The attached log doesn't indicate any issues in Moonraker.  The issue should be resolved on Moonraker's side, however it also requires updates to the frontend.   While I believe that Mainsail has implemented the fix, they have not yet released a new version that includes it.

Arksine, 2024-05-08:
As this discussion has concluded I'm going to close this.

closed 2024-05-08 by Arksine