Arksine/moonraker #813: Need scroll bar for the new "Devices" dialog.

ccatlett1984, 2024-02-17:
### What happened

The current implementation is a huge step forward for ease of use/setup. But, it is not intuitive if you have multiple mcu's (found out by accidently scrolling). I'm suggesting a visible scroll bar, or collapsing sections.

### Client

Mainsail

### Browser

Chrome, Microsoft Edge

### How to reproduce

use the dialog, with multiple MCU's present on the system

Arksine, 2024-02-17:
This appears to be in the wrong repo.  Moonraker doesn't control UI elements, I suspect you will want to provide your feedback to the Mainsail team.

ccatlett1984, 2024-02-17:
Yes, apologies. Shouldn't open issues when I'm tired....

closed 2024-05-08 by Arksine