Arksine/moonraker #814: Z-offset recording

nachidel, 2024-02-17:
### What happened

Hi,

i have a problem for probe calibration.
i did a video for explain : 
[https://drive.google.com/file/d/1crhJ5haMSQxbbxhZbneg6fN3j4x0mq7W/view?usp=drive_link](url)

thanks for your help

### Client

Fluidd

### Browser

Chrome

### How to reproduce

Do the probe calibration...

Arksine, 2024-02-17:
I suspect this is in the incorrect repo.  If you need help with Probe Calibration I recommend asking on the [Klipper Discourse](https://klipper.discourse.group/new).

nachidel, 2024-02-18:
The problem appeared after updating moonraker.  Klipper has not been updated.  The z-offset calibration worked before.  Could you give me some ideas on how to debug this?

Arksine, 2024-05-08:
I hope you were able to get assistance at the recommended link or elsewhere.  As there have been no further updates to this issue, I'm going to close it.  Thanks.

closed 2024-05-08 by Arksine