Arksine/moonraker #818: MQTT Sensors Not Receiving Data

dkaser, 2024-02-21:
### What happened

I have configured a MQTT server and associated sensors in my moonraker configuration. However, the sensors do not receive data.

Symptoms:
- MQTT data is not displayed in the frontend (Fluidd).
- MQTT measurements are not available via the Moonraker API:
```
curl http://127.0.0.1:7125/server/sensors/info?sensor=esp32_temp
{"result": {"id": "esp32_temp", "friendly_name": "Enclosure Temperature", "type": "mqtt", "values": {}}}
```
```
curl http://127.0.0.1:7125/server/sensors/measurements?sensor=esp32_temp
{"result": {"esp32_temp": {}}}
```
Troubleshooting performed:
- Verified that moonraker is connected to the broker (EMQX).
    - moonraker states are being published to the broker
    - EMQX shows that the moonraker client is subscribed to the sensor (esphome/3d-printer/sensor/3d_printer_light_bme280_temperature/state
- Verified that EMQX is sending state updates for the sensor
    - "TCP Packets Sent" for the moonraker client increases at the frequency of the sensor.
- Verified that the moonraker client has permission to the MQTT topic
    - Configured separate MQTT client with identical settings (server, port, credentials, MQTT version, subscriptions). This MQTT client shows the state updates.
    - Compared the standalone client to the moonraker client in EMQX, they appear identical aside from client ID/IP.
- Tried MQTT protocol v3.1.1 and v5; both experience the problem.

### Client

Mainsail, Fluidd, Other

### Browser

Other or N/A

### How to reproduce

1. Configure MQTT and sensor (see sample configuration below)
2. Attempt to view sensor data via frontend or API

### Additional information

MQTT/sensor configuration:
```
[mqtt]
address: a.b.c.d
port: 1883
username: {secrets.mqtt.username}
password: {secrets.mqtt.password}
enable_moonraker_api: False
instance_name: moonraker
publish_split_status: True
mqtt_protocol: v5
status_objects:
  webhooks
  extruder
  heater_bed
  print_stats

[sensor esp32_temp]
type: mqtt
name: Enclosure Temperature
state_topic: esphome/3d-printer/sensor/3d_printer_light_bme280_temperature/state
```

dkaser, 2024-02-24:
OK. I tried that and it worked.

It might be worth clarifying in the documentation that state_response_template is a required parameter. The MQTT topics that I was collecting are numeric, so I interpreted this statement as meaning that I didn't need the option:

> The default is the payload.

In contrast, state_topic says:

> This parameter must be provided.

Arksine, 2024-02-24:
You are correct here, there shouldn't be a default and I missed that when I reviewed the contribution.  I have made this change in 0850c16b00052b817126f27af22dba004a0c7366.

closed 2024-03-09 by Arksine