Arksine/moonraker #821: Bug: Component [update_manager] failed post init

Neko-vecter, 2024-02-24:
### What happened

New install use KIAUH "Component [update_manager] failed post init"

Log
``` shell 
2024-02-24 04:01:03,329 [server.py:_initialize_component()] - Performing Component Post Init: [update_manager]
2024-02-24 04:01:03,333 [server.py:_initialize_component()] - Performing Component Post Init: [history]
2024-02-24 04:01:03,834 [base_deploy.py:log_info()] - PackageDeploy: PackageKit Provider Configured
2024-02-24 04:01:03,837 [server.py:_initialize_component()] - Component [update_manager] failed post init
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/server.py", line 247, in _initialize_component
    await ret
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 189, in component_init
    await updater.initialize()
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 51, in initialize
    self._is_valid = storage["is_valid"]
                     ~~~~~~~^^^^^^^^^^^^
KeyError: 'is_valid'
```

### Client

Mainsail, Fluidd

### Browser

Chrome, Safari, Microsoft Edge

### How to reproduce

I am using KIAUH to install klipper moonraker and webui (mainsail or fluidd). when install finished the [update_manager] can't init with the refresh button.

if use update_manager update to this version it fine.

moonraker commit id `119f579a44ab569524cbae12852362fc8ece1e68`

### Additional information

moonraker.conf
```
[server]
host: 0.0.0.0
port: 7125
klippy_uds_address: /home/pi/printer_data/comms/klippy.sock

[authorization]
trusted_clients:
    10.31.0.0/16
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128
cors_domains:
    http://*.lan
    http://*.local
    https://my.mainsail.xyz
    http://my.mainsail.xyz
    https://app.fluidd.xyz
    http://app.fluidd.xyz
[octoprint_compat]

[history]

[update_manager]
channel: dev
refresh_interval: 168

[update_manager mainsail]
type: web
channel: stable
repo: mainsail-crew/mainsail
path: ~/mainsail
```

Neko-vecter, 2024-02-24:
update

the fix? 

change `update_manager.py` line 51 `storage["is_valid"]` to `self.repo.is_valid()`
``` python3
async def initialize(self) -> Dict[str, Any]:
    storage = await super().initialize()
    await self.repo.restore_state(storage)
    self._is_valid = self.repo.is_valid()
    if not self.needs_refresh():
        self.repo.log_repo_info()
    return storage
```

Still need more test? But it work.

Arksine, 2024-02-24:
This should be resolved in 90e13c07f7430680ba3581a94ec325dab851e60d.

Arksine, 2024-02-24:
Issue has been confirmed as resolved in #822, thanks.

closed 2024-02-24 by Arksine