Arksine/moonraker #822: failed to detect branch when switching from dirty repo to new fork of plugin

sprior, 2024-02-24:
### What happened

With a fully up to date Klipper  v0.12.0-114-ga77d0790/Moonraker v0.8.0-315-g119f579a/Fluidd v1.28.1

I've been trying to modify the source for klippain-shaketune to see if I can make it work with python 3.11 installed from source.  Initially I installed from the original git repo and modified some files and update manager showed status as dirty.  So I tried to go legit and fork the git repo.  After making the appropriate install.sh changes to point to my fork I've installed and doing a git status in ~/klippain_shaketune indicates a clean repo.  I confirmed that the update manager entry in moonraker.conf points to my fork so everything should be good, but update manager still shows:
Klippain-ShakeTune
?
Failed to detect git branch 
Failed to detect repo url 

And in my moonraker.log I'm seeing:
2024-02-23 23:25:17,119 [server.py:_initialize_component()] - Component [update_manager] failed post init
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/server.py", line 247, in _initialize_component
    await ret
  File "/home/pi/moonraker/moonraker/components/update_manager/update_manager.py", line 189, in component_init
    await updater.initialize()
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 51, in initialize
    self._is_valid = storage["is_valid"]
KeyError: 'is_valid'

Which I see in other issues like https://github.com/Arksine/moonraker/issues/821

I also found:
https://github.com/Arksine/moonraker/issues/820
which talks about a stale state loading from storage, but that issue is closed

Is there something I can do to manually clear out an old state from storage when I switch to a fork?

### Client

Fluidd

### Browser

Firefox

### How to reproduce

See description, I think it's all there.

Arksine, 2024-02-24:
Thanks.  This should be resolved in 90e13c07f7430680ba3581a94ec325dab851e60d

Arksine, 2024-02-24:
Issue confirmed resolved, so I will close this.

closed 2024-02-24 by Arksine

sprior, 2024-02-24:
Confirmed, thanks for the quick fix!