Arksine/moonraker #823: Startup failed: NameError: name 'botocore' is not defined

DavidCarl, 2024-02-26:
### What happened

Installed moonraker through KIUAH and when trying to startup moonraker the following error is generated. Its a totally clean install, I installed klipper before moonraker, but thats it.

Its on Debian11 (CB1 image)

```
2024-02-26 09:14:17,316 [server.py:load_component()] - Component (secrets) loaded
2024-02-26 09:14:17,332 [server.py:load_component()] - Component (template) loaded
2024-02-26 09:14:17,353 [server.py:load_component()] - Component (klippy_connection) loaded
2024-02-26 09:14:18,121 [server.py:load_component()] - Unable to load component: (application)
Traceback (most recent call last):
  File "/home/biqu/moonraker/moonraker/server.py", line 290, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/biqu/moonraker/moonraker/components/application.py", line 37, in <module>
    from streaming_form_data import StreamingFormDataParser, ParseFailedException
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/streaming_form_data/__init__.py", line 1, in <module>
    from streaming_form_data.parser import (  # NOQA
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/streaming_form_data/parser.py", line 4, in <module>
    from streaming_form_data._parser import ErrorGroup, _Parser  # type: ignore
  File "streaming_form_data/_parser.pyx", line 6, in init streaming_form_data._parser
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/streaming_form_data/targets.py", line 3, in <module>
    import smart_open  # type: ignore
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/__init__.py", line 34, in <module>
    from .smart_open_lib import open, parse_uri, smart_open, register_compressor  # noqa: E402
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/smart_open_lib.py", line 35, in <module>
    from smart_open import doctools
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/doctools.py", line 21, in <module>
    from . import transport
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/transport.py", line 104, in <module>
    register_transport("smart_open.s3")
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/transport.py", line 49, in register_transport
    submodule = importlib.import_module(submodule)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/s3.py", line 121, in <module>
    RETRY = Retry()
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/s3.py", line 86, in __init__
    self.exceptions: List[Exception] = [botocore.exceptions.EndpointConnectionError]
NameError: name 'botocore' is not defined
2024-02-26 09:14:18,122 [server.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/home/biqu/moonraker/moonraker/server.py", line 655, in main
    server = Server(app_args, log_manager, event_loop)
  File "/home/biqu/moonraker/moonraker/server.py", line 104, in __init__
    self.moonraker_app: MoonrakerApp = self.load_component(config, "application")
  File "/home/biqu/moonraker/moonraker/server.py", line 290, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/biqu/moonraker/moonraker/components/application.py", line 37, in <module>
    from streaming_form_data import StreamingFormDataParser, ParseFailedException
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/streaming_form_data/__init__.py", line 1, in <module>
    from streaming_form_data.parser import (  # NOQA
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/streaming_form_data/parser.py", line 4, in <module>
    from streaming_form_data._parser import ErrorGroup, _Parser  # type: ignore
  File "streaming_form_data/_parser.pyx", line 6, in init streaming_form_data._parser
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/streaming_form_data/targets.py", line 3, in <module>
    import smart_open  # type: ignore
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/__init__.py", line 34, in <module>
    from .smart_open_lib import open, parse_uri, smart_open, register_compressor  # noqa: E402
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/smart_open_lib.py", line 35, in <module>
    from smart_open import doctools
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/doctools.py", line 21, in <module>
    from . import transport
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/transport.py", line 104, in <module>
    register_transport("smart_open.s3")
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/transport.py", line 49, in register_transport
    submodule = importlib.import_module(submodule)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/s3.py", line 121, in <module>
    RETRY = Retry()
  File "/home/biqu/moonraker-env/lib/python3.9/site-packages/smart_open/s3.py", line 86, in __init__
    self.exceptions: List[Exception] = [botocore.exceptions.EndpointConnectionError]
NameError: name 'botocore' is not defined
2024-02-26 09:14:18,124 [server.py:main()] - Server Shutdown
```

### Client

Other

### Browser

Other or N/A

### How to reproduce

Not sure other than installing the newest version

DavidCarl, 2024-02-26:
I guess there is some sort of breaking changes in an underlying library.. Sadly I am not able to look into this myself atm, but will try later.

WickedInvi, 2024-02-26:
For me was after update. Here is my log too if it helps

```2024-02-26 10:14:56,563 [server.py:load_component()] - Component (secrets) loaded
2024-02-26 10:14:56,565 [server.py:load_component()] - Component (template) loaded
2024-02-26 10:14:56,568 [server.py:load_component()] - Component (klippy_connection) loaded
2024-02-26 10:14:56,691 [server.py:load_component()] - Unable to load component: (application)
Traceback (most recent call last):
  File "/home/deyan/moonraker/moonraker/server.py", line 290, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/deyan/moonraker/moonraker/components/application.py", line 37, in <module>
    from streaming_form_data import StreamingFormDataParser, ParseFailedException
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/streaming_form_data/__init__.py", line 1, in <module>
    from streaming_form_data.parser import (  # NOQA
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/streaming_form_data/parser.py", line 4, in <module>
    from streaming_form_data._parser import ErrorGroup, _Parser  # type: ignore
  File "streaming_form_data/_parser.pyx", line 6, in init streaming_form_data._parser
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/streaming_form_data/targets.py", line 3, in <module>
    import smart_open  # type: ignore
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/__init__.py", line 34, in <module>
    from .smart_open_lib import open, parse_uri, smart_open, register_compressor  # noqa: E402
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/smart_open_lib.py", line 35, in <module>
    from smart_open import doctools
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/doctools.py", line 21, in <module>
    from . import transport
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/transport.py", line 104, in <module>
    register_transport("smart_open.s3")
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/transport.py", line 49, in register_transport
    submodule = importlib.import_module(submodule)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/s3.py", line 121, in <module>
    RETRY = Retry()
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/s3.py", line 86, in __init__
    self.exceptions: List[Exception] = [botocore.exceptions.EndpointConnectionError]
NameError: name 'botocore' is not defined
2024-02-26 10:14:56,691 [server.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/home/deyan/moonraker/moonraker/server.py", line 655, in main
    server = Server(app_args, log_manager, event_loop)
  File "/home/deyan/moonraker/moonraker/server.py", line 104, in __init__
    self.moonraker_app: MoonrakerApp = self.load_component(config, "application")
  File "/home/deyan/moonraker/moonraker/server.py", line 290, in load_component
    module = importlib.import_module(full_name)
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 883, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/deyan/moonraker/moonraker/components/application.py", line 37, in <module>
    from streaming_form_data import StreamingFormDataParser, ParseFailedException
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/streaming_form_data/__init__.py", line 1, in <module>
    from streaming_form_data.parser import (  # NOQA
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/streaming_form_data/parser.py", line 4, in <module>
    from streaming_form_data._parser import ErrorGroup, _Parser  # type: ignore
  File "streaming_form_data/_parser.pyx", line 6, in init streaming_form_data._parser
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/streaming_form_data/targets.py", line 3, in <module>
    import smart_open  # type: ignore
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/__init__.py", line 34, in <module>
    from .smart_open_lib import open, parse_uri, smart_open, register_compressor  # noqa: E402
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/smart_open_lib.py", line 35, in <module>
    from smart_open import doctools
[... cut ...]
  File "/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/s3.py", line 121, in <module>
    RETRY = Retry()
  File "/home/deyan/moonraker-env/lib/python3.10/site-packages/smart_open/s3.py", line 86, in __init__
    self.exceptions: List[Exception] = [botocore.exceptions.EndpointConnectionError]
NameError: name 'botocore' is not defined
2024-02-26 10:14:56,691 [server.py:main()] - Server Shutdown
```

Arksine, 2024-02-26:
Please provide full logs as requested when creating an issue.  This looks like a dependency issue, but I need all details by the log to have an idea as to what the underlying problem is.

xyleth, 2024-02-26:
I'm having the same issue.  If I've read the logs right the problem is with the sqlite_packages module smart_open component which seems to be trying to load a S3 adapter.  Which makes sense why botocore is being called.

For now I just activated the moonraker pyenv and did `pip install boto3` and now Moonraker starts again.

Arksine, 2024-02-26:
Thanks.  The issue is with an upstream dependency, specifically related to https://github.com/piskvorky/smart_open/issues/804.

Moonraker doesn't depend on `smart_open` directly, however `streaming-form-data` pulls it in.  Unfortunately SFD doesn't pin to a specific version, and 7.0 of `smart_open` has a bug.  I'm going to attempt to temporarily pin smart open in Moonraker's requirements as a workaround.

Arksine, 2024-02-26:
Workaround pushed in 9813011db2e65f13072b8e31ad9a9364e99e338d.  New installations should no longer attempt to pull in smart_open, 7.0.  If you have a broken installation, the following should fix it:
```
~/moonraker-env/bin/pip install smart_open<=6.4.0
```

DavidCarl, 2024-02-26:
Solved, pulled master and reinstalled. Thanks @Arksine

@WickedInvi solution is here.

closed 2024-02-26 by DavidCarl