Arksine/moonraker #827: Moonraker Error with some Filenames

icebear74, 2024-03-06:
### What happened

Moonraker cant load Preview with some Filenames

### Client

Fluidd

### Browser

Chrome, Firefox

### How to reproduce

[image]
Filename: #12 baby dragon_PLA_3h1m.gcode
seems the Spaces are the Problem :)

Arksine, 2024-03-06:
There is no log attached, so I can't see the exact error.  FWIW, my test client handles file names with spaces ok.  I suspect that the spaces in the file name need to be escaped by the front end.

Arksine, 2024-05-08:
As there have been no further comments or updates to this issue I'm going to close this.

closed 2024-05-08 by Arksine