Arksine/moonraker #828: Failed to load button

Zilli341, 2024-03-06:
### What happened

I'm trying to add a button to toggle a solid-state relay but it's not working. I get the following error:
 "_Failed to load button [button PowerButton]
[button PowerButton]: Option 'pin' encountered the following error while parsing: [Errno 6] Opening input line handle: No such device or address_" 

Using the same GPIO pin as an output works, and I can read it fine with gpiod. Changing pin doesn't do anything.
I'm on an OrangePI Prime running Armbian Bookworm and everything should be up to date.

### Client

Mainsail, Fluidd, KlipperScreen

### Browser

Firefox, Other or N/A

### How to reproduce

Add a [button] section and set a pin. 

### Additional information

I was messing around with other stuff, the error should be at the end of the file.
[moonraker(3).log](https://github.com/Arksine/moonraker/files/14513930/moonraker.3.log)

Arksine, 2024-03-06:
Typically these kind of errors are system related, ie: an issue with permissions, the kernel build, etc.

When you state that you can read it using `gpiod`, do you mean using `gpiomon`?  What version of `gpiod` is on the system?  You can get it by running:

```
gpiodetect --version
````

Zilli341, 2024-03-07:
I always used gpioget as a quick check to see if it was working, but trying to use gpiomon returns the following error: "gpiomon: error waiting for events: No such device".
I guess you were right and there is a problem somewhere deeper in the system and not in the application.

`gpiodetect (libgpiod) v1.6.3
Copyright (C) 2017-2018 Bartosz Golaszewski
License: LGPLv2.1
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
`

Zilli341, 2024-03-07:
I figured it out. In the end, it was something I probably should have noticed sooner. 
The Allwinner H5 doesn't support interrupts on the PC1-16 GPIOs, which are mostly located on the right side of the connector. I always chose random pins from that side when testing, so it never worked. 
After a few Google searches and a closer look at the datasheet, I figured out that was probably the reason. I switched to PAxx GPIOs, and now it's working.

Thank you for the help.

Arksine, 2024-03-09:
Thanks for the feedback.  Glad you were able to fix it.

closed 2024-03-09 by Arksine