Arksine/moonraker #832: Device Power on mqtt zigbee2mqtt payload

Travis90x, 2024-03-25:
### What happened

In according to the [MQTT Device Configuration](url=https://moonraker.readthedocs.io/en/latest/configuration/#mqtt-device-configuration)

I correctly manage a tasmota over mqtt with this
```
[power MQTT]
type: mqtt
command_topic: cmnd/Printer_Name/POWER 
query_topic: cmnd/Printer_Name/POWER
state_topic: stat/Printer_Name/POWER
command_payload:
    {command}
state_response_template:
    {payload}
query_after_command: False
```
So the Power On command is 
cmnd/Printer_Name/POWER On

But I cant' manage a zigbee2mqtt device that needs the command:
zigbee2mqtt/Friendly_name/get payload {"state": "on"}
or
zigbee2mqtt/Friendly_name/set/state on

I set in my moonraker.conf

```
[mqtt]
address: 192.168.1.2 # mqtt borker IP
port: 1883
username: mqtt_user
password: my_password
mqtt_protocol: v3.1.1
enable_moonraker_api: True
  
[power MQTT]
type: mqtt
command_topic: zigbee2mqtt/Friendly_name/set/state
command_payload:
    {command}

query_topic: zigbee2mqtt/Friendly_name  # the correct command should be zigbee2mqtt/Friendly_name  /get payload {"state": ""}

state_topic: zigbee2mqtt/Friendly_name/get/state
state_response_template:
     # {payload}  # I no need entire payload, only the "state" value
    {% set resp = payload|fromjson %}
    { resp["state"] }
query_after_command: true
```

Command_topic won't send commands to MQTT broker

If I set query_topic: zigbee2mqtt/Friendly_name/set/state 
Log will be: Zigbee2MQTT:error 2024-03-26 00:26:05: Publish 'set' 'state' to 'Friendly_name' failed: 'Error: '' not allowed, choose between: toggle,off,on
because "zigbee2mqtt/Friendly_name/set/state" send command without On/Off/Toggle,
Ok, anyway query_topic send commands to MQTT broker.

Power Device wont send mqtt command written in command_topic & command_payload.
Only query_topic: send commands but wrongly, it should send

I tried:
zigbee2mqtt/Friendly_name
zigbee2mqtt/Friendly_name/set
zigbee2mqtt/Friendly_name/set/state
zigbee2mqtt/Friendly_name/get
zigbee2mqtt/Friendly_name/get/state

in vary combination with
command_topic, query_topic and state_topic

with no results...

### Client

Mainsail

### Browser

Firefox

### How to reproduce

zigbee2mqtt/Friendly_name/set {command} is wrong

zigbee2mqtt need zigbee2mqtt/Friendly_name/set {  "state": "ON" }
or 
zigbee2mqtt/Friendly_name/set/state on

### Additional information

[image]

Arksine, 2024-03-26:
Without `moonraker.log` it isn't clear what is happening with the command topic, but I suspect the primary issue is that the device can't initialize and thus commands won't be sent.  If you need to send a payload with the query topic you need to configure the `query_payload` option (see the [docs](https://moonraker.readthedocs.io/en/latest/configuration/#mqtt-device-configuration).  You can use Jinja2 to encode a dict to json if that is what is required for your zigbee client.  Moonraker has some brief [Jinja2 docs](https://moonraker.readthedocs.io/en/latest/configuration/#jinja2-templates), and the full Jinja template documentation is available [here](https://jinja.palletsprojects.com/en/3.1.x/templates/).

Travis90x, 2024-03-26:
Thank for your reply! Uploaded below moonraker.log and klippy.log with no usefull log

Ok, if I set 
command_topic: zigbee2mqtt/Friendly_name/set/state
command_payload:  {command}
**query_topic: zigbee2mqtt/Friendly_name/set/state 
query_payload: toggle**
The Zigbee2mqtt device toggle in loop ON-OFF every 5 seconds because of query_topic that works as SET and not GET the state

But the Power button doesn't work, so I suppose that is necessary to configure correctly the query_topic that it's needed to enable command_topic

I tested with MQTTX (client mqtt)
```
zigbee2mqtt/Friendly_name/get
zigbee2mqtt/Friendly_name/get state
zigbee2mqtt/Friendly_name/get {"state": ""}
```
give me
`Zigbee2MQTT:error 2024-03-26 22:51:08: Invalid message 'null', skipping...`

If use
`zigbee2mqtt/Friendly_name/get/state`
I obtain

`Zigbee2MQTT:info  2024-03-26 22:52:12: MQTT publish: topic 'zigbee2mqtt/Friendly_name', payload '{"current":0,"energy":0,"indicator_mode":"off/on","linkquality":232,"over_current_breaker":"ON","over_current_threshold":65,"over_voltage_breaker":"ON","over_voltage_threshold":265,"power":0,"power_breaker":"ON","power_outage_memory":"restore","power_threshold":13,"state":"ON","temperature":26.94,"temperature_breaker":"ON","temperature_threshold":100,"under_voltage_breaker":"ON","under_voltage_threshold":75,"update":{"installed_version":-1,"latest_version":-1,"s`

So I don't need query_payload, right?

Otherwise I can use it with
`zigbee2mqtt/Friendly_name/get {"state": ""}`
I obtain the payload with MQTTX
but If I set
```
query_topic: zigbee2mqtt/Friendly_name/get 
query_payload: {"state": ""}
```
**The Power Button Disappear** because the correct Jinja2 output shoud be 
query_payload: {'{"state": ""}'}

Correct it, Power still no work

I try 
query_topic: zigbee2mqtt/Friendly_name/get/state
query_payload: state
MQTTX obtain payload, But the Power button still no work

If I set 
state_topic: zigbee2mqtt/Friendly_name/set/state
state_response_template: toggle
Power button disappear

state_topic: zigbee2mqtt/Friendly_name/get/state
state_response_template: state
Power Button goes dark

###################################
###################################
_For my reminder_
**Button Name (MQTT) goes dark:**
###################################
state_topic: zigbee2mqtt/Friendly_name/get
state_response_template: {'{"state": ""}'}
query_topic: zigbee2mqtt/Friendly_name/get
query_payload: {'{"state": ""}'}
###################################
state_topic: zigbee2mqtt/Friendly_name/get/state
state_response_template: state
      {% set resp = payload|fromjson %}
      { resp["state"]|lower }
 ###################################    
state_topic: zigbee2mqtt/Friendly_name/get/state
state_response_template: state
###################################
state_topic: zigbee2mqtt/Friendly_name/get/state
state_response_template:
###################################
state_topic: zigbee2mqtt/Friendly_name/get/state
state_response_template: 
query_topic: zigbee2mqtt/Friendly_name/get/state
query_payload: 
###################################
state_topic: zigbee2mqtt/Friendly_name/get/state
state_response_template: state
query_topic: zigbee2mqtt/Friendly_name/get/state
query_payload: state
###################################
state_topic: zigbee2mqtt/Friendly_name/get
state_response_template:
    {% set resp = payload|fromjson %}
    { resp["state"]|lower }
query_topic: zigbee2mqtt/Friendly_name/get
query_payload: {'{"state": ""}'}
###################################

###################################
###################################
**Button Name (MQTT) goes light:**
###################################
state_topic: zigbee2mqtt/Friendly_name/get
state_response_template: {payload}
###################################
state_topic: zigbee2mqtt/Friendly_name/set
state_response_template: state
###################################
state_topic: zigbee2mqtt/Friendly_name/get
state_response_template:
    {% set resp = payload|fromjson %}
    { resp["state"]|lower }
query_topic: zigbee2mqtt/Friendly_name/get/state
query_payload: state
###################################
state_topic: zigbee2mqtt/Friendly_name/get
state_response_template:
    {% set resp = payload|fromjson %}
    { resp["state"]|lower }
query_topic: zigbee2mqtt/Friendly_name/get/state
query_payload: {'{"state": ""}'}

[klippy.log](https://github.com/Arksine/moonraker/files/14765386/klippy.log)
[moonraker.log](https://github.com/Arksine/moonraker/files/14765387/moonraker.log)
[image]

Arksine, 2024-03-26:
Based on the info you provided, it looks like the state topic needs to be `zigbee2mqtt/Friendly_name`.  The state topic is the topic Moonraker subscribes to and listens for updates.  The "zigbee2mqtt" client should publish all updates to this topic.  If the zigbee sets the retain flag on this topic then the `query_topic` is optional.

Travis90x, 2024-03-27:
Now I'm configuring the powermeter function with

[sensor mqtt_powermeter]
type: mqtt
name: Powermeter
state_topic: zigbee2mqtt/Friendly_name
state_response_template:
  {% set notification = payload|fromjson %}
  {set_result("power", notification["apower"]|float)}
  {set_result("voltage", notification["voltage"]|float)}
  {set_result("current", notification["current"]|float)}
  
  But the documentation is poor to understand how show in Mainsail those sensor values.

Arksine, 2024-03-27:
Unfortunately I can't help with setting up Mainsail, as its a separate project.  Moonraker just exposes the APIs which frontend applications like Mainsail can choose to implement.  I don't know if Mainsail can show data from generic sensors or how to configure it if it can.

Arksine, 2024-05-08:
As the original issue has been resolved I'm going to close this.

closed 2024-05-08 by Arksine