Arksine/moonraker #837: Device locked_while_printing: True asserting while offline

keyboarderror, 2024-04-04:
### What happened

In the most recent v0.8.0-325-g99b97afc update locked_while_printing: True is asserting at startup by default. I could not power on my printer. Commenting it out (or setting False) restored access to the power button.

My configuration:

[power Printer]
type: gpio
pin: gpio17
locked_while_printing: True
on_when_job_queued: True
restart_klipper_when_powered: True
restart_delay: 2
bound_service: klipper
initial_state: off
off_when_shutdown: True
off_when_shutdown_delay: 180

### Client

Mainsail

### Browser

Chrome

### How to reproduce

locked_while_printing: True

Arksine, 2024-04-05:
It is necessary to attach `moonraker.log` reproducing the issue for me to confirm the bug and track it down.  Thanks.

keyboarderror, 2024-04-05:
It is appended under additional information but I don't see it.

keyboarderror, 2024-04-05:
Trying as an attachment.
[moonraker.log](https://github.com/Arksine/moonraker/files/14878789/moonraker.log)

Arksine, 2024-04-05:
The log does not contain a traceback indicating that a request to toggle power was denied.    This suggests that the frontend is denying request.

keyboarderror, 2024-04-20:
Closing as the most recent update seems to have resolved the problem.

closed 2024-04-20 by keyboarderror